Sloeber/arduino-eclipse-plugin #783: Leonardo upload still having problems (but is already much better)

MAWMN, 2017-07-08:
Hello,
  here is the Leonardo guy again, with more information. With the nightly build the upload is already going much better, but 2 out of 10 times still going wrong.

**This is the error I receive:**
java.lang.NullPointerException
	at java.util.regex.Matcher.getTextLength(Unknown Source)
	at java.util.regex.Matcher.reset(Unknown Source)
	at java.util.regex.Matcher.<init>(Unknown Source)
	at java.util.regex.Pattern.matcher(Unknown Source)
	at jssc.SerialPortList.getWindowsPortNames(SerialPortList.java:311)
	at jssc.SerialPortList.getPortNames(SerialPortList.java:294)
	at jssc.SerialPortList.getPortNames(SerialPortList.java:178)
	at io.sloeber.core.api.Serial.list(Serial.java:132)
	at io.sloeber.core.communication.ArduinoSerial.wait_for_com_Port_to_appear(ArduinoSerial.java:74)
	at io.sloeber.core.communication.ArduinoSerial.makeArduinoUploadready(ArduinoSerial.java:217)
	at io.sloeber.core.tools.uploaders.arduinoUploader.uploadUsingPreferences(arduinoUploader.java:58)
	at io.sloeber.core.tools.uploaders.UploadSketchWrapper$UploadJobWrapper.run(UploadSketchWrapper.java:178)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)

**And this is a copy of the console at that moment:**
Starting upload
using arduino loader
Forcing reset using 1200bps open/close on portCOM10
PORTS { CNCA0, CNCB0, COM1, COM7, COM8, COM10, COM25,} / { config error:, see https://github.com/jantje/arduino-eclipse-plugin/issues/252,} => { config error:, see https://github.com/jantje/arduino-eclipse-plugin/issues/252,}
Comport reset took250ms
Using comportconfig error: from now onwards
Ending reset

LaunchingE:\Eclipse_Sloeber_Arduino\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -CE:\Eclipse_Sloeber_Arduino\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/etc/avrdude.conf -patmega32u4 -cavr109 -Pconfig error: -b57600 -D -Uflash:w:E:\Eclipse_Sloeber_WS\First_Test/Release/First_Test.hex:i 
Output:
avrdude: ser_open(): can't open device "config": Het systeem kan het opgegeven bestand niet vinden.

avrdude done.  Thank you.

/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude finished
upload done

**This is a copy of the console of the Arduino IDE:**
Een reset wordt geforceerd (met 1200bps openen/sluiten) op poort COM10
PORTS {CNCA0, CNCB0, COM1, COM7, COM8, COM10, COM25, } / {CNCA0, CNCB0, COM1, COM7, COM8, COM25, } => {}
PORTS {CNCA0, CNCB0, COM1, COM7, COM8, COM25, } / {CNCA0, CNCB0, COM1, COM7, COM8, COM9, COM25, } => {COM9, }
Found upload port: COM9
E:\Tools\Electronica\arduino-1.6.11\hardware\tools\avr/bin/avrdude -CE:\Tools\Electronica\arduino-1.6.11\hardware\tools\avr/etc/avrdude.conf -v -patmega32u4 -cavr109 -PCOM9 -b57600 -D -Uflash:w:C:\Users\Marcel\AppData\Local\Temp\build0cd86ac184092c3731ad72037cfae4f0.tmp/Blink.ino.hex:i 

avrdude: Version 6.0.1, compiled on Apr 15 2015 at 19:59:58
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "E:\Tools\Electronica\arduino-1.6.11\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM9
         Using Programmer              : avr109
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega32U4
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : butterfly
         Description     : Atmel AppNote AVR109 Boot Loader

Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

avrdude: devcode selected: 0x44
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% -0.00s

avrdude: Device signature = 0x1e9587
avrdude: reading input file "C:\Users\Marcel\AppData\Local\Temp\build0cd86ac184092c3731ad72037cfae4f0.tmp/Blink.ino.hex"
avrdude: writing flash (4148 bytes):

Writing | ################################################## | 100% 0.32s

avrdude: 4148 bytes of flash written
avrdude: verifying flash memory against C:\Users\Marcel\AppData\Local\Temp\build0cd86ac184092c3731ad72037cfae4f0.tmp/Blink.ino.hex:
avrdude: load data flash data from input file C:\Users\Marcel\AppData\Local\Temp\build0cd86ac184092c3731ad72037cfae4f0.tmp/Blink.ino.hex:
avrdude: input file C:\Users\Marcel\AppData\Local\Temp\build0cd86ac184092c3731ad72037cfae4f0.tmp/Blink.ino.hex contains 4148 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.04s

avrdude: verifying ...
avrdude: 4148 bytes of flash verified

**Having a look at the command for AVR dude:**
_**Eclipse IDE:**_
E:\Eclipse_Sloeber_Arduino\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -CE:\Eclipse_Sloeber_Arduino\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/etc/avrdude.conf -patmega32u4 -cavr109 -Pconfig error: -b57600 -D -Uflash:w:E:\Eclipse_Sloeber_WS\First_Test/Release/First_Test.hex:i 

_**Arduino IDE**_
E:\Tools\Electronica\arduino-1.6.11\hardware\tools\avr/bin/avrdude -CE:\Tools\Electronica\arduino-1.6.11\hardware\tools\avr/etc/avrdude.conf -v -patmega32u4 -cavr109 -PCOM9 -b57600 -D -Uflash:w:C:\Users\Marcel\AppData\Local\Temp\build0cd86ac184092c3731ad72037cfae4f0.tmp/Blink.ino.hex:i 

==> Look at the part that says **avrdude.conf -v -patmega32u4** of the Arduino IDE..

I think this is where the problem is...

The Eclipse IDE (or AVRDUDE) seems to be too fast or soon trying to send the data to the Arduino (COM9 not attached yet).... 
(See also the ERROR reported : **Forcing reset using 1200bps open/close on portCOM10
PORTS { CNCA0, CNCB0, COM1, COM7, COM8, COM10, COM25,} / { config error:, see https://github.com/jantje/arduino-eclipse-plugin/issues/252,} => { config error:, see https://github.com/jantje/arduino-eclipse-plugin/issues/252,}
Comport reset took250ms** )

Probably the Arduino IDE gets just a little bit time to have the PC Switched and Attached the COM9 again, probably because of this -v  (Verbose) output..

Hope this helps to improve further..

kind regards,

Marcel

jantje, 2017-07-08:
The problem for avrdude is caused by
`Using comportconfig error: from now onwards`
Which you also see in the avrdude command as
`-Pconfig`
The problem is caused by jssc when asked the list of com ports.
This was reported in #252. In those days we got a uncaught exception. So for #252 I catch the exception and report this is the exception additionally caught in #255. (fortunately as I already forgot)
Only 2 people reported this issue so something "special" (read on your system) is at play here.
I do not have this issue here (I just made a upload test board containing a leonardo and it never happened).

MAWMN, 2017-07-08:
Hello,

   I have already solved the problem myself as I suggested above by adding the extra verbose (-v)  parameter.
The score is now 0 errors out of 10 tries, and this over and over again.. ;-)

**This is what I have done:**
[image]

However what I have seen is when I create a new project I have to change this parameter again for every single project I create.
Is there a solution to make this global ??

regards,

Marcel

jantje, 2017-07-08:
The -v is for verbose uploading. It doesn't make sense to me that this fixes the problem. (I'm not saying it doesn't)
If it does it would mean the leonardo is in a different state after the upload that makes the next reset work "better"
Anyways: If it fixes the problem you can add the -v to the platform.txt. This will make it do so for all new/modified projects

MAWMN, 2017-07-09:
Hello,

    today added the -v to the platform.txt

Works like a charm...NO more errors because the COM port has not appeared yet..

The extra console output it creates now, probably slows down the application just a couple of milliseconds on my system, just enough to have the new COM port being available again..

I am closing the subject now..

Again thans for the support and info..

Marcel

jantje, 2017-07-09:
Cool I'm happy there is a workaround.
Thx for the feedback

jantje, 2017-07-09:
I reopen as a workaround is not a real fix. I fear I'll have to look at jssc for this and other issues

jantje, 2018-05-16:
I  just finished quite a bit of stuff in the  upload area.
Could you give this a new test?

closed 2019-01-09 by jantje