Sloeber/arduino-eclipse-plugin #784: Error: Program "make" not found in PATH

garyrice, 2017-07-11:
I downloaded the Sloeber full product install file to my Mac. It unpacked successfully and appeared to complete the installation process successfully as well.

I then opened an existing Arduino sketch from my Arduino folder. I selected "Verify" from the Arduino menu.

The console window showed:
Error: Program "make" not found in PATH

I did a full Finder search for make.exe. No such file exists on my computer. What gives?

Gary

jantje, 2017-07-11:
Make does not come with sloeber on mac. For details on how to get make (and other mac specific stuff) see here: http://eclipse.baeyens.it/installAdvice.shtml

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

garyrice, 2017-07-11:
I reviewed your suggested link. There were 2 possibiliries:
1) xcode (which I don't have)
2) Arduino IDE prior to v1.5.7

I downloaded v1.5 but it contained no separate make.app file - just the IDE executable as a monolithic file.

Other suggestions for finding make?

jantje, 2017-07-11:
You should be able to open the monolitic app in something like the file manager

garyrice, 2017-07-12:
I was able to locate a copy of "make" in an old version of xcode. I installed the file fragment into Sloeber using the path that was referenced in the error message. Doing so allowed me to step further along the way to using Sloeber. The next error is:
make: *** No rule to make target  `/Users/gary/Desktop/Files', needed by `Lathe_Box.elf'.  Stop.

Ideas?

Gary

jantje, 2017-07-12:
Have you watched the video's? This path looks very wierd

garyrice, 2017-07-12:
>This path looks very wierd

Now I remember reading that path names must NOT contain blanks. So I replaced the blanks  (5 instances) with underline characters.

YAA!! Sloeber is now working on my Mac!

Thanks for your help.

Gary

closed 2017-07-12 by jantje