Sloeber/arduino-eclipse-plugin #786: Unable to make or compile on fresh installation

carlfeld, 2017-07-13:
Hi! 
This is messages from a clean installation ( latest version 4.1 on windows 10 ), Made new project, with  bare minimum example:

BareMinimum.ino

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

Messages I get:  seems IDE does not build an adecuate MAKE file.]

10:24:21 **** Clean-only build of configuration Release for project aaz ****
"C:\\sloeber\\arduinoPlugin\\tools\\make\\make" clean 
make: *** No rule to make target 'clean'.  Stop.

10:24:24 Build Finished (took 3s.845ms)

10:24:25 **** Build of configuration Release for project aaz ****
"C:\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\.ino.cpp'
'Starting C++ compile'
"C:\sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_DUEMILANOVE -DARDUINO_ARCH_AVR   -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\variants\standard" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\.ino.cpp"  -o  ".ino.cpp.o"
'Finished building: ..\.ino.cpp'
' '
make: *** No rule to make target 'feldman\Documents\sloeber-workspace\aaz\Release\arduino.ar', needed by 'aaz.elf'.  Stop.

10:24:27 Build Finished (took 1s.820ms)

jantje, 2017-07-13:
Have you waited for the installation to complete before creating the project?
What is in the core and variant folder of the project?

carlfeld, 2017-07-13:
hi, of course I waited the system to  install all the additional software.
I selected Arduino 2009 ( it's the card I own ) with atmega 328, Also tried Arduino NG with Atmega 8 ( I want to adapt a frequency measurement library to his processor ) , and the contents of core and variant are the ussual files you may thing:
May be I need to try another Arduino version.

jantje, 2017-07-13:
What is in the core and variant folder of the project?
Unlikely a other version will make a difference. ,This is basic functionality that should work.

carlfeld, 2017-07-14:
sorry, I tryed changing the card to arduino uno and atmega, and it was the same thing. 
Then deleted the \sloeber directory, and the workplace directory, downloaded version 4.0, unziped it in the sloeber directory, let it upgrade files, and it was the same stuff, with a simple empty arduino project.
The only thing I can say about my machine ( I5, Windows 10 32 bit, 3GB ram ) is that it has no serial port.
Any advise of what to look for ?

jantje, 2017-07-14:
for the third time: What is in the core and variant folder of the project?

carlfeld, 2017-07-16:
thanks a lot for your time and effort.I had to solve a problem related to changing a processor family in a code with tons of conditional compilation, so Arduino wasn't helping me, so finally did it at Atmel studio, and left sloeber for another time.
Core and variant seems to be the regular ones in arduino 1.67 standard, Anyway the problem detected seems to be with a make unable to compile or even make clean.
As I told you, this was in a particular windows 10 32 bits I5 computer, when I also run Atmel Studio 7 , Arduino 8  and Mplab for microchip, on a fresh installation .
Thanks for your time  again.

closed 2017-07-16 by carlfeld