Sloeber/arduino-eclipse-plugin #788: double #includes and #defines

uzi18, 2017-07-24:
@jantje : solution from #775 works like a charm and it is compatible with Arduino IDE.
First time I'm able to try Sloeber :)

Do You remember why sloeber.ino.cpp is not simple file with just include of project.ino?
Why it redefines all #define from ino file and double include all #include "something.h"?
This behavior breaks compilation process
Is it a bug?

jantje, 2017-07-24:
The .ino format requires these additional defenitions.
If you do not want then just use .cpp files

uzi18, 2017-07-24:
Can't confirm that.
10 minutes ago have reduced sloeber.ino.cpp to
```
#include <Arduino.h>
#include "project.ino"
```
launched make project.elf (in Release dir) do the trick, so why we need all of it?

closed 2017-07-24 by jantje

uzi18, 2017-07-24:
I see no discussion here.

jantje, 2017-07-24:
Let me rephrase "The .ino format requires these additional definitions."
The ino format does not enforce definitions before usage like c++ does.
So for gcc to be able to process .ino files definitions need to be added.
Because it is easier to add all definitions 'whether they are needed or not" Sloeber adds all definitions it can handle.

uzi18, 2017-07-24:
I just think about solution to process ino files in the same behavior like
Arduino builder.
As simple as posible,  less problem in the future.

Some defines are wrong:
Don't do double include of local files with #include "something.h"
And don't do double #defines as gcc rises redefine errors

You know what I mean?

If You want I could prepare minimal test source.

jantje, 2017-07-24:
>Don't do double include of local files with #include "something.h"

That is why I added the option to add pragma once to header file
 
>And don't do double #defines as gcc rises redefine errors

Where I tested double defines do not cause problems if the defines are the same value.
I tested with over 1000 examples of libraries and the current implementation works fine.
Don't see the need to change it now.

uzi18, 2017-07-24:
I will try to prepare minimal example for unit tests.