Sloeber/arduino-eclipse-plugin #793: Trouble Compiling Sparkfun SAMD21 Board

mikekpat, 2017-07-26:
I have been using the eclipse IDE with the Sloeber arduino plugin for some time.

I was interested in implementing the sparkfun wireless joystick project but have
run into problems getting the Sparkfun SAMD21 Development Board to compile.
I provide the most simple pin13 blink program as a test. Then attempt to compile.
The compiler fails two things. The first one "missing sam.h" I managed to fix by
supplying an additional include to an CMSIS-Atmel directly.

But now I'm getting stumped on the second compile fail of  USB. The error messages
are:

core/core/USB/subdir.mk:45: recipe for target 'core/USBCore.cpp.o' failed
make: *** [core/USBCore.cpp.o] Error 1

The errors have the following form:
apps/eclipse-neon3/arduinoPlugin/packages/SparkFun/hardware/samd/1.2.0/cores/arduino/USB/SAMD21_USBDevice.h:94:96: error: 'volatile struct USB_DEVICE_EPINTFLAG_Type::<anonymous>' has no member named 'STALL'
  inline bool epBank0IsStalled(ep_t ep)           { return usb.DeviceEndpoint[ep].EPINTFLAG.bit.STALL & 1; }
 
So, it's complaining about missing definitions. I found a usb.h file than seems to define the missing member constants,
but all attempts to include paths to the location of the usb.h file do nothing to resolve the problem.

So, while the Arduino Zero board compiles fine in eclipse IDE, the Sparkfun samd21 fails.
Can you suggest anything to resolve this compile problem?

jantje, 2017-07-27:
It compiles fine on my system.
Are you on the latest nightly? I have changed a lot in this area lately

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

mikekpat, 2017-07-27:
Thanks for replying so quickly. I chose the latest version (1.6.9) of Sparkfun SAMD Board from the Arduino->Preferences->Platforms and Boards window and that fixed it.

I don't see any Sloeber plugin that is newer than what I have installed, even though I installed back in May 2017. Eclipse reports Sloeber version 4.1.0.201705252335. I had installed the IDE and Sloeber as a single install bundle. Is there something newer? The Sloeber web page seems to only show 4.0 versions available.

closed 2017-07-27 by mikekpat

jantje, 2017-07-27:
If you activate the sloeber nightly update site in windows->preferences->update and updtae->available update sites and run help->update you should get a newer version