Sloeber/arduino-eclipse-plugin #794: process_begin: CreateProcess(NULL, C:/cccc/workspace/kolp/Release/arduino.ar failed

fozc, 2017-07-28:
Hello everyone,
First off all, thanks for the plugin@jantje.

I downloaded the 64 bits bundle package from http://eclipse.baeyens.it, unzip the C:\slaeber.
After that the IDE downloaded the arduino things. But I cant compile any thing. 

I tried to add "cmd.exe /c" but it didnt works.here is my error logs.

 
[image]

14:37:38 **** Build of configuration Release for project kolp ****
"C:\sloeber\arduinoPlugin\tools\make\make" all
'Building file: ...ino.cpp'
'Starting C++ compile'
I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\variants\standard" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "...ino.cpp" ".ino.cpp.o"
Dosya ad?, dizin ad? veya birim etiketi s”zdizimi hatal?.(The file name, directory name, or volume label syntax is incorrect)
subdir.mk:24: recipe for target '.ino.cpp.o' failed
make: [.ino.cpp.o] Error 1 (ignored)
'Finished building: ...ino.cpp'
' '
'Building file: C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\CDC.cpp'
'Starting C++ compile'
I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\variants\standard" -MMD -MP -MF"core\CDC.cpp.d" -MT"core\CDC.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\CDC.cpp" "core\CDC.cpp.o"
Dosya ad?, dizin ad? veya birim etiketi s”zdizimi hatal?. (The file name, directory name, or volume label syntax is incorrect)
core\core\subdir.mk:98: recipe for target 'core\CDC.cpp.o' failed
make: [core\CDC.cpp.o] Error 1 (ignored)
'Finished building: C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\CDC.cpp'
' '
'Building file: C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\HardwareSerial.cpp'
'Starting C++ compile'
I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\variants\standard" -MMD -MP -MF"core\HardwareSerial.cpp.d" -MT"core\HardwareSerial.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\HardwareSerial.cpp" "core\HardwareSerial.cpp.o"
Dosya ad?, dizin ad? veya birim etiketi s”zdizimi hatal?.
core\core\subdir.mk:105: recipe for target 'core\HardwareSerial.cpp.o' failed
make: [core\HardwareSerial.cpp.o] Error 1 (ignored)
'Finished building: C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\HardwareSerial.cpp'
' '
'Building file: C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\HardwareSerial0.cpp'
'Starting C++ compile'
I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\variants\standard" -MMD -MP -MF"core\HardwareSerial0.cpp.d" -MT"core\HardwareSerial0.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\HardwareSerial0.cpp" "core\HardwareSerial0.cpp.o"
Dosya ad?, dizin ad? veya birim etiketi s”zdizimi hatal?.
core\core\subdir.mk:112: recipe for target 'core\HardwareSerial0.cpp.o' failed
make: [core\HardwareSerial0.cpp.o] Error 1 (ignored)
'Finished building: C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\HardwareSerial0.cpp'
'
...................................
......There are more same things..
...................................

'Building file: C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\wiring_shift.c'
'Starting C compile'
I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\variants\standard" -MMD -MP -MF"core\wiring_shift.c.d" -MT"core\wiring_shift.c.o" -D__IN_ECLIPSE__=1 "C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\wiring_shift.c" "core\wiring_shift.c.o"
Dosya ad?, dizin ad? veya birim etiketi s”zdizimi hatal?.
core\core\subdir.mk:266: recipe for target 'core\wiring_shift.c.o' failed
make: [core\wiring_shift.c.o] Error 1 (ignored)
'Finished building: C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\cores\arduino\wiring_shift.c'
' '
'Starting archiver'
C:/cccc/workspace/kolp/Release/arduino.ar .\core\CDC.cpp.o .\core\HardwareSerial.cpp.o .\core\HardwareSerial0.cpp.o .\core\HardwareSerial1.cpp.o .\core\HardwareSerial2.cpp.o .\core\HardwareSerial3.cpp.o .\core\IPAddress.cpp.o .\core\PluggableUSB.cpp.o .\core\Print.cpp.o .\core\Stream.cpp.o .\core\Tone.cpp.o .\core\USBCore.cpp.o .\core\WInterrupts.c.o .\core\WMath.cpp.o .\core\WString.cpp.o .\core\abi.cpp.o .\core\hooks.c.o .\core\main.cpp.o .\core\new.cpp.o .\core\wiring.c.o .\core\wiring_analog.c.o .\core\wiring_digital.c.o .\core\wiring_pulse.S.o .\core\wiring_pulse.c.o .\core\wiring_shift.c.o
makefile:73: recipe for target 'C:\cccc\workspace\kolp\Release\arduino.ar' failed
process_begin: CreateProcess(NULL, C:/cccc/workspace/kolp/Release/arduino.ar .\core\CDC.cpp.o .\core\HardwareSerial.cpp.o .\core\HardwareSerial0.cpp.o .\core\HardwareSerial1.cpp.o .\core\HardwareSerial2.cpp.o .\core\HardwareSerial3.cpp.o .\core\IPAddress.cpp.o .\core\PluggableUSB.cpp.o .\core\Print.cpp.o .\core\Stream.cpp.o .\core\Tone.cpp.o .\core\USBCore.cpp.o .\core\WInterrupts.c.o .\core\WMath.cpp.o .\core\WString.cpp.o .\core\abi.cpp.o .\core\hooks.c.o .\core\main.cpp.o .\core\new.cpp.o .\core\wiring.c.o .\core\wiring_analog.c.o .\core\wiring_digital.c.o .\core\wiring_pulse.S.o .\core\wiring_pulse.c.o .\core\wiring_shift.c.o, ...) failed.
make (e=2): Sistem belirtilen dosyayı bulamıyor.

make: *** [C:\cccc\workspace\kolp\Release\arduino.ar] Error 2

14:37:43 Build Finished (took 4s.365ms)

"-E" not found in PATH

What is your suggestions ? Thank you very much.

jantje, 2017-07-28:
I see a couple of very strange things
First of: normally the build stops after the first failing file. here he goes on. Which is a specific CDT setting which does not match your comment "After that the IDE downloaded the arduino things. But I cant compile any thing."
Secondly the command is missing for the cpp compile and for the archiver.
There must be some timing issue (should be fixed after doing project properties->arduino->apply->ok)
Or a download extract issue (check the content of C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.19\platform.txt)
or something else

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

fozc, 2017-08-05:
Hi jante,

Firstly, I tried to "project properties->arduino->apply->ok" but nothing changed. .I checked the platform.txt an it seems okey. I tried to 32 bit bundle, no luck. Even another computer I get the same result. I think , I am missing something.

İs only 64/32 bit bundle setup and java enough for compile ? Do I need CDT setup ?
Thank you.

jantje, 2017-08-10:
I install the product in a couple of video's on my youtube channel

jantje, 2017-09-01:
@fozc 
Is there a update on this issue? Can it be closed?

jantje, 2017-09-11:
closing due to no updates

closed 2017-09-11 by jantje