Sloeber/arduino-eclipse-plugin #798: Wrong port used for uploading on Micro board

auron-muc, 2017-08-04:
I run Sloeber 4.1.0.201705252335 on windows, and recently (4.1?) the upload for a project on arduino micro fails (worked before but I haven't uploaded that project in a while). 

It is correcly detecting the boot loader port (COM5) but still connects to the regular port COM7 in avrdude. Since COM7 is gone at that time avrdude fails.

Anything I mis-configured or is this a regression?

Thanks,
Tobias

########### upload session ##################
Starting upload
using arduino loader
Forcing reset using 1200bps open/close on portCOM7
PORTS { COM3, COM4, COM7, COM8, COM11,} / { COM3, COM4, COM7, COM8, COM11,} => {}
PORTS { COM3, COM4, COM7, COM8, COM11,} / { COM3, COM4, COM5, COM7, COM8, COM11,} => { COM5,}
Comport reset took500ms
Using comportCOM5 from now onwards
Ending reset

LaunchingC:\sloeber-ide\sloeber_41\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -CC:\sloeber-ide\sloeber_41\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/etc/avrdude.conf -patmega32u4 -cavr109 -PCOM7 -b57600 -D -Uflash:w:C:\Users\Tobias\JoystickIntercept/Release/JoystickIntercept.hex:i 
Output:
avrdude: ser_open(): can't open device "\\.\COM7": Das System kann die angegebene Datei nicht finden.

avrdude done.  Thank you.

/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude finished
upload done

auron-muc, 2017-08-04:
Ah, I currently recover by copying the command line from the output and pasting it in a CMD shell (with fixed port). Obviously this is not very conveniant. ;-)

auron-muc, 2017-08-06:
I checked the 4.1 branch and in arduinoUploader.java:62 an environment entry for Const.ENV_KEY_SERIAL_PORT_FILE is created ( but not for Const.ENV_KEY_SERIAL_PORT).

Yet my [Project->Properties->c/C++ Build->Build Variables -> show system variables] "A.TOOLS.AVRDUDE.UPLOAD.PATTERN" was 
`"${A.TOOLS.AVRDUDE.CMD.PATH}" "-C${A.TOOLS.AVRDUDE.CONFIG.PATH}" ${A.TOOLS.AVRDUDE.UPLOAD.VERBOSE} ${A.TOOLS.AVRDUDE.UPLOAD.VERIFY} -p${A.BUILD.MCU} -c${A.UPLOAD.PROTOCOL} -P${A.SERIAL.PORT} -b${A.UPLOAD.SPEED} -D "-Uflash:w:${A.BUILD.PATH}/${A.BUILD.PROJECT_NAME}.hex:i"`

Changing this to 
`"${A.TOOLS.AVRDUDE.CMD.PATH}" "-C${A.TOOLS.AVRDUDE.CONFIG.PATH}" ${A.TOOLS.AVRDUDE.UPLOAD.VERBOSE} ${A.TOOLS.AVRDUDE.UPLOAD.VERIFY} -p${A.BUILD.MCU} -c${A.UPLOAD.PROTOCOL} -P${A.SERIAL.PORT.FILE} -b${A.UPLOAD.SPEED} -D "-Uflash:w:${A.BUILD.PATH}/${A.BUILD.PROJECT_NAME}.hex:i"`

in the Arduino micro project fixed it for me. ( note the ".FILE" in "-P${A.SERIAL.PORT.FILE}" )

jantje, 2017-08-09:
I think this is fixed in the nightly

auron-muc, 2017-08-15:
Verified working in nightly as of 2017-08-09.
Since the workaround for 4.1 is also documented I'll close this.

closed 2017-08-15 by auron-muc

auron-muc, 2017-08-31:
@jantje Maybe you should leave the issue open until the next release, so people can find it more easily?
The mindframe of people might be: closed means fixed **and released**. I personally also took a look in the closed issues, but that is a habit due to my daytime work I think ( if people search before posting an issue at all *sigh*) 

P.S. A also saw the other leonardo issue, but figured it was a different topic.

jantje, 2017-08-31:
@auron-muc 
That was how we worked. However having lots of issues open to find the real open issues is not appreciated by developers. As the developers are more supporting than the users I opted to start closing the issues. Keeping issues open is also "non-standard" in open source world so the tools do not cater for this behaviour so Sloeber gets bad scores because of long lasting open issues.
 
To "soften the pain" I made a "do I need to upgrade" menu item in Sloeber. The idea is that this page shows you what has changed in which version since the version you are running.
Unfortunately the page needs some TLC because it still dumps to many issue and looks like a developer did the layout ;-)

Please feel free to update the page https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/website/WebContent/github_status.php so filter and make it look better.

auron-muc, 2017-08-31:
@jantje I can have a look, but I fear the "like a developer did the layout" thing will get worse :)

jantje, 2017-08-31:
@auron-muc I'm a functional guy. I find the filter more important