Sloeber/arduino-eclipse-plugin #800: V4.1 : config starts at every startup and esp8266 is not in hardware list

ghislainborremans, 2017-08-06:
Hello
I used Sloeber up to 2 months ago without problems but after the  install 4.1 i no longer have the esp8266 anymore . Also, every time i start up, the config starts ( building the arduinoplugin ) and stops because of error : Failed to configer Sloeber
Failed to extract tar.bz2.D:/sloeber/arduinoPlugin/downloads/examples_Arduino_1_6_7.zip
Can't extract fileD:\sloeber\arduinoPlugin\examples\01.Basics\AnalogReadSerial\AnalogReadSerial.ino, file already exists!

I suppose the configuration does not finish well and so the esp8266 is never installed?? 

thanks in advance

ghislainborremans, 2017-08-06:
Hello 
i redownloaded again the nightly and now the config ran completely. 
i added the third partuy json of esp8266 and now all is ok.

closed 2017-08-06 by ghislainborremans

jantje, 2017-08-09:
as you know there are download issues with esp8266 so I removed it from the default list.
We also fixed the "download json files at startup" which didn't work and which causes the "configuration"