Sloeber/arduino-eclipse-plugin #803: Failed to run the Upload recipe

agomas, 2017-08-28:
Hi,

I'm trying to upload a sketch I'm getting this error:

eclipse.buildId=unknown
java.version=1.8.0_131
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=pt_PT
Framework arguments:  -perspective io.sloeber.application.perspective -perspective io.sloeber.application.perspective
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -perspective io.sloeber.application.perspective -data file:/E:/sloeber-workspace/ -perspective io.sloeber.application.perspective

io.sloeber.arduino.core
Error
Fri Aug 25 15:57:24 BST 2017
Failed to run the Upload recipe 

java.io.IOException: Cannot run program "" (in directory "E:\sloeber-workspace"): CreateProcess error=87, O parâmetro está incorreto
	at java.lang.ProcessBuilder.start(Unknown Source)
	at io.sloeber.core.tools.ExternalCommandLauncher.launch(ExternalCommandLauncher.java:294)
	at io.sloeber.core.tools.uploaders.GenericLocalUploader.RunConsoledCommand(GenericLocalUploader.java:40)
	at io.sloeber.core.tools.uploaders.arduinoUploader.uploadUsingPreferences(arduinoUploader.java:82)
	at io.sloeber.core.tools.uploaders.UploadSketchWrapper$UploadJobWrapper.run(UploadSketchWrapper.java:177)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)
Caused by: java.io.IOException: CreateProcess error=87, O parâmetro está incorreto
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(Unknown Source)
	at java.lang.ProcessImpl.start(Unknown Source)
	... 6 more

Follows the properties:
[image]

I'm using the bluepill. I have now problems in a computer with an earlier installation, but now I'm trying to install all the environment to program the bluepill in other computers and the error is always the same.

I have tried the other solutions in the issue #714, but the error is the same.

What is missing?

Thank you in advance.

Regards

jantje, 2017-08-28:
what does the upload console say?

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

agomas, 2017-08-28:
The console only have this:

Starting upload
using arduino loader
Starting reset using DTR toggle process
Toggling DTR
Continuing to useCOM5
Ending reset
failed to upload

*
*Regards,*
André Gomes*
Imagem intercalada 1
*T:*+351 963 292 674
*S:* www.atronia-tech.pt <http://www.atronia-tech.pt>

---
Este e-mail foi verificado em termos de vÃ­rus pelo AVG.
http://www.avg.com

jantje, 2017-08-28:
you are probably missing environment variables because the platform is not installed by the boards manager included in sloeber.

agomas, 2017-08-29:
I still have the same problem, when install the platatform by the boards manager.

[image]

jantje, 2017-08-29:
I tried it here and the linking fails in the nightly.
I will take a look later today

jantje, 2017-08-30:
I tried it here with the nightly.
I found a bug related to the library search and already fixed that one.
I also found the linking didn't work due to hard coded paths which I fixed locally.
Once that was fixed the upload command is launched (but failed due to missing boards)
I'm currently running some unit tests and when these are successful I'll release the fix and you should be able to get it tomorrow.

jantje, 2017-08-31:
I tried to test the changes but apparently the https://raw.githubusercontent.com/stm32duino/BoardManagerFiles/master/STM32/package_stm_index.json file has been updated and your platform has disappeared. :-(

agomas, 2017-08-31:
Thanks for the effort to correct the bugs.

There are other json that support the bluepill.
With the bugs fixed, I should be able program the board. I suppose.

I will try it.

jantje, 2017-08-31:
txs keep us posted

jantje, 2017-09-11:
@agomas Any update here?

agomas, 2017-09-13:
Now I'm able to program, but with problems with the libraries.
Probably the problem is in the json that I'm using.

I will have to change to a faster processor. Is slober compatible with the Teensy 3.6 board?

jantje, 2017-09-13:
>I will have to change to a faster processor. Is slober compatible with the Teensy 3.6 board?

yes

jantje, 2017-09-13:
So we can close this issue?

agomas, 2017-09-14:
Yes.

Thank you for the support. 
I am going to "became a patron". What you have build is remarkable and give us a lot of flexibility to program in arduino.

Regards

closed 2017-09-14 by jantje