Sloeber/arduino-eclipse-plugin #804: Build Issue

shayneloyd, 2017-08-29:
The Backstory:
I recently installed Eclipse Oxygen(from website) and Sloeber (through eclipse market place) on Ubuntu 17.04. I really would like to get Eclipse w/ Sloeber working as it seems to offer a lot more in an IDE than the Arduino IDE. I have separated my project files into .cpp and . h files as that appears to be the preferred method. I did have an issue when I first installed Sloeber that the avr-gcc was not there, so I created the requested directory and put the tool chain from my Arduino 1.8.4 installation there. I created if I remember the entire tools directory. I have since uninstalled and reinstalled Sloeber through the eclipse IDE, but I'm not sure if it changed this.

The Issue:

```
"/opt/eclipse//arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-gcc" -c -g -Os -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -flto -fno-fat-lto-objects -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   -I"/opt/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.20/cores/arduino" -I"/opt/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.20/variants/standard" -I"/home/shayne/Arduino/libraries/MCP_2515" -I"/opt/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.20/libraries/SPI" -I"/opt/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.20/libraries/SPI/src" -MMD -MP -MF"core/WInterrupts.c.d" -MT"core/WInterrupts.c.o" -D__IN_ECLIPSE__=1 "/opt/eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.20/cores/arduino/WInterrupts.c"  -o  "core/WInterrupts.c.o"

cc1: error: -fno-fat-lto-objects are supported only with linker plugin

core/core/subdir.mk:182: recipe for target 'core/WInterrupts.c.o' failed

make: *** [core/WInterrupts.c.o] Error 1

```
I'm not sure what this message means except it does not like the -fno-fat-lto-objects

I also get the popup error message:

```
ArchiveNameProvider: The call should go to the overloaded function not here.
LinkNameProvider: The call should go to the overloaded function not here.

```
In addition:
I am using an Interrupt in my program as I noticed it stops on the WInterrupts.c.o

In Conclusion:
I'm sure it is something; I'm doing wrong. I am new to Eclipse and don't understand all the options as most of my projects have been done in the simple Arduino IDE. Any help is appreciated. Please let me know if you would like any further information.

Thank you,
Shayne

P.S.  I am going to build a simple project to attempt to isolate the issue.

jantje, 2017-08-29:
As a newbe I strongly advice to install the product. That contains eclipse and the plugin. 
However what you are seeing is a real compilation issue so it looks as if the installation is ok. Though  you should not have to create directories and copy stuff from arduino. So it may be a installation problem after all.
Have you seen the installation video's?

shayneloyd, 2017-08-30:
I have watched some different videos about eclipse/sloeber, but not the those. I found the download link on the website and got it running. It is still giving me errors, but now they are error messages telling me; I don't know how to program. I'm sure the problem was my self installed avr-gcc folder. Thank you for your help.

jantje, 2017-08-30:
If I understand correctly the issue is solved and we can close  the issue.
If I understood wrongly please reopen.

closed 2017-08-30 by jantje