Sloeber/arduino-eclipse-plugin #807: Leonardo/Pro Micro upload doesn't use the right serial port on Windows 7

pierre-quelin, 2017-08-31:
Env.
Sloeber 4.1.0.201705252335
Sloeber, the Eclipse Arduino IDE 4.0.0.201705251445
Windows 7 Pro 64bits

When I try to upload my sketch, resetting the board works fine.
The plugin waits and detects the new serial port correctly.
But when avrdude is launch it doesn't use it ! It uses the old one.
```
Starting upload
using arduino loader
Forcing reset using 1200bps open/close on portCOM9
PORTS { COM9,} / { COM9,} => {}
PORTS { COM9,} / { COM8,} => { COM8,}
Comport reset took500ms
Using comportCOM8 from now onwards
Ending reset

LaunchingC:\Solystic\sloeber\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -CC:\Solystic\sloeber\/arduinoPlugin/packages/SparkFun/hardware/avr/1.1.6/avrdude.conf -patmega32u4 -cavr109 -PCOM9 -b57600 -D -Uflash:w:C:\Users\quelin\Documents\sloeber-workspace\test/Release/test.hex:i 
Output:
avrdude: ser_open(): can't open device "\\.\COM9": Le fichier spécifié est introuvable.

avrdude done.  Thank you.

/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude finished
upload done
```
When I replace the variable A.TOOLS.AVRDUDE.UPLOAD.PATTERN with the appropriate serial port (-PCOM8), everything works fine.
"${A.TOOLS.AVRDUDE.CMD.PATH}" "-C${A.TOOLS.AVRDUDE.CONFIG.PATH}" ${A.TOOLS.AVRDUDE.UPLOAD.VERBOSE} -p${A.BUILD.MCU} -c${A.UPLOAD.PROTOCOL} -PCOM8 -b${A.UPLOAD.SPEED} -D "-Uflash:w:${A.BUILD.PATH}/${A.BUILD.PROJECT_NAME}.hex:i"

auron-muc, 2017-08-31:
see https://github.com/Sloeber/arduino-eclipse-plugin/issues/798

pierre-quelin, 2017-08-31:
Seems to works like a charme.
I'm sorry.
Best regards

closed 2017-08-31 by jantje