Sloeber/arduino-eclipse-plugin #809: macOS:  "Sloeber" is damaged and can't be opened.  You should move it to the Trash.

jimconrad, 2017-09-06:
An attempt to execute sloeber by double-clicking the sloeber icon after expanding the macOS product archive, V4.1_mac64.2017-05-25_16-52-16.tar.gz, results in, "sloeber" is damaged and can't be opened.  You should move it to the Trash.  Please see the attached screenshot.

[image]

I've reproduced the problem on two different macs and two browsers (for downloading), safari and firefox, and two different underlying networks, using OS X 10.12.6.

Please contact me if I can be of assistance.  Jim Conrad.  Boise State University.  Boise, Idaho, USA.

mjmeijer, 2017-09-06:
The solution is in https://github.com/Sloeber/arduino-eclipse-plugin/issues/625
Though we still haven't updated the website yet, sorry

jimconrad, 2017-09-06:
Thanks and sorry about the duplicate.   I may have time to update the site later this week.

closed 2017-09-06 by jimconrad