Sloeber/arduino-eclipse-plugin #810: upload fails for ATTinyCore ATtiny85 & ATtiny84

liebman, 2017-09-06:
When avrdude is run it does not pass a value for the -c option and its passing an empty -b option.

```
Launching/Users/liebman/Applications/cpp-neon/Eclipse.app/Contents/Eclipse//arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -C/Users/liebman/Arduino/hardware/ATTinyCore/avr/avrdude.conf -pattiny85 -c -P/dev/cu.wchusbserial1d133440 -b -D -Uflash:w:/Users/liebman/src/AnalogClock/I2CAnalogClock/Release/I2CAnalogClock.hex:i 
Output:
avrdude: invalid baud rate specified '-D'
```

While the arduino IDE runs the following (note the -c option and that it does not include any -P or -b options)

```
/Users/liebman/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -C/Users/liebman/Documents/Arduino/hardware/ATTinyCore/avr/avrdude.conf -v -pattiny85 -cusbtiny -Uflash:w:/var/folders/6x/wzh7qty17t5dzbc_z9rgl1z00000gn/T/arduino_build_723413/PWMTest.ino.hex:i 
```

This was its current (github) ATTinyCore and the latest nightly:
[image]

liebman, 2017-09-06:
Looking at platform.txt it defines multiple commands "upload", "program", "erase" and "bootloader".  For my case I'm using a programmer not a serial upload and ideally sloeber would use the "program" command.

> # AVR Uploader/Programmers tools
> # ------------------------------
> tools.avrdude.path={runtime.tools.avrdude.path}
> tools.avrdude.cmd.path={path}/bin/avrdude
> tools.avrdude.config.path={runtime.platform.path}/avrdude.conf
> 
> 
> tools.avrdude.upload.params.verbose=-v
> tools.avrdude.upload.params.quiet=-q -q
> tools.avrdude.upload.pattern="{cmd.path}" "-C{config.path}" {upload.verbose} -p{build.mcu} -c{upload.protocol} -P{serial.port} -b{upload.speed} -D "-Uflash:w:{build.path}/{build.project_name}.hex:i"
> 
> tools.avrdude.program.params.verbose=-v
> tools.avrdude.program.params.quiet=-q -q
> tools.avrdude.program.pattern="{cmd.path}" "-C{config.path}" {program.verbose} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{build.path}/{build.project_name}.hex:i"
> 
> tools.avrdude.erase.params.verbose=-v
> tools.avrdude.erase.params.quiet=-q -q
> tools.avrdude.erase.pattern="{cmd.path}" "-C{config.path}" {erase.verbose} -p{build.mcu} -c{protocol} {program.extra_params} -e -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:{bootloader.high_fuses}:m -Ulfuse:w:{bootloader.low_fuses}:m
> 
> tools.avrdude.bootloader.params.verbose=-v
> tools.avrdude.bootloader.params.quiet=-q -q
> tools.avrdude.bootloader.pattern="{cmd.path}" "-C{config.path}" {bootloader.verbose} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{runtime.platform.path}/bootloaders/{bootloader.file}:i"
>

jantje, 2017-09-07:
@liebman 
Thanks for being a patron.
To investigate this issue I really need the json file you used to install the platform and a screen dump of your arduino project settings. 
For an example of what is needed to investigate a board specific issue see issue #808 .

As a quick and dirty workaround to use the programmer you can try the following : 
modify the platform.txt to remove the upload command info and rename the programmer to upload:
```
**#**tools.avrdude.upload.params.verbose=-v
**#**tools.avrdude.upload.params.quiet=-q -q
**#**tools.avrdude.upload.pattern="{cmd.path}" "-C{config.path}" {upload.verbose} -p{build.mcu} -c{upload.protocol} -P{serial.port} -b{upload.speed} -D "-Uflash:w:{build.path}/{build.project_name}.hex:i"

tools.avrdude.**upload**.params.verbose=-v
tools.avrdude.**upload**.params.quiet=-q -q
tools.avrdude.**upload**.pattern="{cmd.path}" "-C{config.path}" {**upload*.verbose} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{build.path}/{build.project_name}.hex:i"
```
After having done  the changes go to project properties->arduino->apply->apply and close

liebman, 2017-09-07:
I did not add the core via a "Third party index url's".  I checked out the git version in ~/Arduino/hardware and that makes the code show up as an option in the project properties:
[image]

liebman, 2017-09-07:
If one wanted to install via JSON its 'http://drazzy.com/package_drazzy.com_index.json'

jantje, 2017-09-08:
The json has the same issue.
I'm looking for the root cause now.

jantje, 2017-09-08:
Arduino IDE puts
`-cstk500v1 ...  -b19200` 
I searched but I have no clue where Arduino IDE gets this info from

jantje, 2017-09-08:
I made a quick fix hard coding default upload.protocol and upload.speed.
I think the package creator should have a look because they should know where this value  comes from.

As to selecting the programmer command. can you do that in the Arduino IDE? and if so how?

liebman, 2017-09-08:
Select from Tools->Programmer as shown.  Then click the upload button as normal.

[image]

jantje, 2017-09-09:
The default upload is not working as it should be. I hardcoded the values fir this board and I hope the  issue above can help remove the hard coding.
But the programmer does not seem to pick up the right values. I'm investigating.

jantje, 2017-09-09:
This :looks better :-)
`D:\eclipse\sloeber\eclipse\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -CD:\eclipse\sloeber\eclipse\/arduinoPlugin/packages/ATTinyCore/hardware/avr/1.1.4/avrdude.conf -pattiny85 -cusbtiny -Uflash:w:D:\workspaces\runtime-arduino.product\testtt/Release/testtt.hex:i`

jantje, 2017-09-11:
Can you check the nightly? I think this one is fixed.

liebman, 2017-09-11:
I'll give it a try when I get home from work.

liebman, 2017-09-11:
Yes! Works now!

jantje, 2017-09-11:
good :-)
Thanks for reporting.

closed 2017-09-11 by jantje