Sloeber/arduino-eclipse-plugin #811: Can't make .hex file

trialabc, 2017-09-07:
I am new to arduino and I have just found out there is a plugin that bundle them together. Unfortunately, after creating a new sketch and build without error, it has written "make: *** No rule to make target C:\Users\Kenny', needed by 'XXX.elf'. Stop. "
I don't see any .elf file is made and I don't see any .hex file is made. I am not sure why.

jantje, 2017-09-07:
Have you watched some of the video's here? https://www.youtube.com/watch?v=zc2JLbtbI7Q&list=PLCSjf7QmP3TDPSjT_5UIp0DHXzKdhIQsF
I think  `"make: *** No rule to make target C:\Users\Kenny', ` as running clean could delete C:\Users\Kenny.

trialabc, 2017-09-08:
When I do a "clean" operation, it shows the following...

The system cannot find the path specified.
makefile:94: recipe for target 'clean' failed
make: [clean] Error 1 (ignored)

I am using window 10 with eclipse 64-bit Oxygen. Since those makefiles are auto-generated, I am not sure what could have gone wrong....

trialabc, 2017-09-08:
I have just tested once more but this time I am placing my files in a path without spacing. I simply use one of the example "Blink" for Arduino Nano. 
And it WORKS!! I can compile, I can clean and I can upload.....

So I am guessing it is because of the space in my path? 
My "workspace" is placed in C://Users/Kenny Tam/workspace. When compiled, the path contains a space and thus execute wrongly.
Maybe it would be better to have double quotation to the path? Anyway I can change the auto-generated makefiles? If not I will simply move to another path for the time being..

jantje, 2017-09-08:
duplicate of #643

closed 2017-09-11 by jantje