Sloeber/arduino-eclipse-plugin #813: No upload to Arduino Micro - OSX

Dekoninck, 2017-09-08:
I installed Sloeber 4.1 in Eclipse Neon. (4.6.3) on OSX 10.1.2 and uploads are OK to Uno and Micro boards.  On Micro boards uploads fail (board Genuino Micro is set). Upload from Arduino 1.8.4 IDE works fine.

//************************************************************************
Starting upload
using arduino loader
Forcing reset using 1200bps open/close on port/dev/cu.usbmodemFD121
PORTS { /dev/cu.BlueRSI091B-serialport, /dev/cu.usbmodemFD121,} / { /dev/cu.BlueRSI091B-serialport,} => {}
PORTS { /dev/cu.BlueRSI091B-serialport, /dev/cu.usbmodemFD121,} / { /dev/cu.BlueRSI091B-serialport, /dev/cu.usbmodemFD121,} => {}
Comport appeared and disappeared with same name
Using comport/dev/cu.usbmodemFD121 from now onwards
Ending reset

Launching/Applications/Eclipse.app/Contents/Eclipse//arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -C/Applications/Eclipse.app/Contents/Eclipse//arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/etc/avrdude.conf -patmega32u4 -cavr109 -P/dev/cu.usbmodemFD121 -b57600 -D -Uflash:w:/Users/xxxxxx/Documents/Eclipse/Micro_test/Release/Micro_test.hex:i 
Output:

Connecting to programmer: .
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
avrdude: butterfly_recv(): programmer is not responding
Found programmer: Id = "S E���"; type = p
    Software Version = V.�; Hardware Version = v. 
avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: error: programmer did not respond to command: leave prog mode
avrdude: error: programmer did not respond to command: exit bootloader

avrdude done.  Thank you.

avrdude finished
upload done

//************************************************************************
xxxxxx = my user name

jantje, 2017-09-08:
this is probably a duplicate of #798 
You can reopen if it is not fixed in the nightly

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

closed 2017-09-08 by jantje

Dekoninck, 2017-09-08:
Thanks for your quick reply,

Changing ".FILE" in "-P${A.SERIAL.PORT.FILE}"  as in #798 does not solve the problem: with this setting the port is not found at all.

avrdude: ser_open(): can't open device "cu.usbmodemFD121": No such file or directory

With the original setting ( no .FILE) the communication with the board starts but the response seems to be wrong.
Difference Windows / OSX?

Kind regards,

JD

jantje, 2017-09-08:
please upgrade to the nightly. Many things have changed in this area since 4.1.
You can see all things that have been fixed in sloeber->menu->arduino->do I need to upgrade.

Dekoninck, 2017-09-09:
Problem solved ! Now uploading to Arduino Micro.

After updating with the nightly version, I could still not upload to the micro. As shown by the led’s on the micro, the usb was doing something, but did not get an answer from the micro, as I previously suspected.

I removed completely the FTDIUSBserial Kext driver (including the cd /private/var/db/receipts/com.FTDI...) and installed the CH34x serial driver.
https://github.com/adrianmihalko/ch340g-ch34g-ch34x-mac-os-x-driver <https://github.com/adrianmihalko/ch340g-ch34g-ch34x-mac-os-x-driver>

Restarted the Mac and now all boards can communicate / accept uploads!

It’ not clear to me why the Arduino IDE can talk trough the FTDI driver and Sloeber/Eclipse cannot.
Thanks Jantje for the feedback!

jantje, 2017-09-09:
thanks for confirming it works

Dekoninck, 2017-11-01:
Dear Jantje,
For information, and for whom it may help.
- The problem reappeared when uploading: the connection with the micro is made but the data traffic is disturbed.
- After research, it showed that the problem returned after I enabled the bluetooth which is also on the USB bus.
- Turning OFF Bluetooth solved the problem and upload is normal.
- The Arduino IDE does not have a problem with the Bluetooth turned on.

Bluetooth vendor id = 0x5ac, prod id = 0x821a

The problem is not with the UNO.

Greetz
JD

jantje, 2017-11-01:
I assume this is a bluetooth breakout board or shield you are referring to.
It looks like a  timing issue to me unless the avrdude command is different in arduino ide

Dekoninck, 2017-11-02:
I’m referring to the internal bluetooth of the MacBookPro. 
Grtz

jantje, 2017-11-02:
This strengthens my opinion this is a timing issue?

Dekoninck, 2017-11-03:
I’m not sure, it’s hard to tell. With Bluetooth turned on, it takes already 15 sec’s to perform the 1200 bps reset. With BT off, 4 sec’s to perform the complete upload. If the slow response is caused by a timing is not clear to me.
Hardware should be OK since the Arduino IDE has no problem.

Grtz

Dekoninck, 2017-11-03:
Some additional info:

Bluetooth ON:
Connecting to programmer: .
Found programmer: Id = "2017.11"; type = .
    Software Version = 2.0; Hardware Version = 1.7
avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: error: programmer did not respond to command: leave prog mode
avrdude: error: programmer did not respond to command: exit bootloader

Bluetooth OFF:
Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

avrdude: AVR device initialized and ready to accept instructions

Looks like programmer ID is date?

jantje, 2017-11-03:
When the arduino board is reset the bootloader waits a while before starting the sketch and during that time it is listening for a upload. This is why the reset is needed before running the upload.
So if it takes to long to detect the port on the pc the board will be running the sketch and is no longer ready to do a upload.
15 seconds is a long time and may be longer than the time the bootloader is waiting (I have no idea to be honest)

uzi18, 2017-11-03:
> otvgaJpZM4PRAvM>.

Blootooth problem is available in java rxtx library,  there is a fix
somewhere.
It is not existing on linux platform.

jantje, 2017-11-03:
Sloeber no longer uses rxtx (nor does arduino IDE) we both use jssc (sloeber uses version 2.8.0 which is the latest version at the time of writing)