Sloeber/arduino-eclipse-plugin #814: Sketch upload to DigiSpark fails

norman-knott, 2017-09-09:
Hello,

I am trying to use Sloeber for programming a Digispark. I've installed the package `http://digistump.com/package_digistump_index.json` and created a Sketch for:

platform folder: `*/digistump/hardware/avr/1.6.7`
board: `Digispark (Default - 16.5mhz)`
upload protocol: `Micronucleus`

The `verify` works perfectly, but when I try to upload the Sketch it fails with:
> Failed to run the Upload recipe 
Cannot run program "" (in directory "/home/crazydoc/Documents/sloeber-workspace"): error=2, Datei oder Verzeichnis nicht gefunden

jantje, 2017-09-09:
have you tried the  nightly?

norman-knott, 2017-09-13:
Hi,
with the nightly build `io.sloeber.product-4.1.0-20170911.231844-88-linux.gtk.x86_64.tar.gz`
I get a similar error.

> Failed to run the Upload recipe 
Cannot run program "/MEDIA/DATA/CRAZYDOC/PROGRAMS/SLOEBER//ARDUINOPLUGIN/PACKAGES/DIGISTUMP/TOOLS/MICRONUCLEUS/2.0A4/MICRONUCLEUS" (in directory "/home/crazydoc/Documents/sloeber-workspace"): error=2, Datei oder Verzeichnis nicht gefunden

> Starting upload
using arduino loader
>
>
>Launching/MEDIA/DATA/MYUSER/PROGRAMS/SLOEBER//ARDUINOPLUGIN/PACKAGES/DIGISTUMP/TOOLS/MICRONUCLEUS/2.0A4/MICRONUCLEUS 
Output:
failed to upload

The `micronucleus` binary exists and is executable. In this version the only problem might be the capital letters.

jantje, 2017-09-13:
Linux is case sensitive so the capital letters are a problem for sure.
I'l investigate the capitalisation first.

norman-knott, 2017-10-09:
@jantje Do you need more input by me?

jantje, 2017-10-09:
Sorry but I havn't had time to investigate this issue yet.
What you could do is
Upgrade to the latest nightly and see if the issue still exists.
Check project properties->C/C++ build->Environment to see whether the capitalisation is before or after the conversion of .txt files to environment variables

norman-knott, 2017-10-10:
@jantje The path is correct in this screen:
[image]

With build `4.1.0-20171009.224450-107-linux.gtk.x86_64` it is the same. Only capital letters.

jantje, 2017-10-10:
That means the upload command is converted to uppercase on your system.
Strange that is.

jantje, 2017-10-20:
I found on my system that if you use default uploader it works ok.
The reson for the uppercase is because there is a "tools.micronucleus.upload.pattern" but no "tools.micronucleus.program.pattern" which is used when a programmer is used.

jantje, 2017-10-20:
Did you try to use the uploader in the arduino IDE. IMHO the programmers.txt file is simply not correctly formatted and the platform.txt is not containing the right info.

closed 2017-10-26 by jantje