Sloeber/arduino-eclipse-plugin #815: platform.txt and build.txt possibly as variable in preferences

mulderp, 2017-09-09:
Hello,

while trying to build my first blinky sketch, I am a bit confused where Sloebers handles the compiler and upload toolchain.

The build works fine with:

```
"/Users/pmu/eclipse/sloeber.app/Contents/Eclipse//arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-size" -A "/Users/pmu/Documents/sloeber-workspace/simpleblink1/Release/simpleblink1.elf"
/Users/pmu/Documents/sloeber-workspace/simpleblink1/Release/simpleblink1.elf  :

```

But when trying to upload I get:

```
Failed to run the Upload recipe 
Cannot run program "/Users/pmu/eclipse/sloeber.app/Contents/Eclipse//arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude" (in directory "/Users/pmu/Documents/sloeber-workspace"): error=2, No such file or directory
```

So, I was looking for the platform.txt file, and it is here:

pm:sloeber.app pmu$ vim ./Contents/Eclipse/arduinoPlugin/packages/arduino/hardware/avr/1.6.20/platform.txt

Now it's not clear how to best fix the problem. Maybe variables in the Arduino preferences to another platform.txt could help ?

jantje, 2017-09-09:
does "/Users/pmu/eclipse/sloeber.app/Contents/Eclipse//arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude" exist on your system?
if not try to remove and reinstall the arduino avr toolchain in arduino->preferences->arduino->platform and boards

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

mulderp, 2017-09-09:
I think the problem was that the build_and_launch.sh script does not copy/link the avrdude toolchain. So, I did it manually and it worked now. For future uses, it might still be nice to have some way of fine tune the upload command. Not sure what the best way for this is though.

closed 2017-09-09 by mulderp

jantje, 2017-09-09:
The build_and_launch.sh  script is to build  and run from source. It is not what I would advice to get started with.
The avrdude toolchain is installed at the first run of sloeber.