Sloeber/arduino-eclipse-plugin #818: Fail to download examples_Arduino_1_6_7.zip

ouryperd, 2017-09-12:
I get the following message after install:

_Failed to configer Sloeber
Failed to extract tar.bz2./home/user/sloeber/arduinoPlugin/downloads/examples_Arduino_1_6_7.zip
Can't extract file/home/user/sloeber/arduinoPlugin/examples/01.Basics/AnalogReadSerial/AnalogReadSerial.ino, file already exists!_

I deleted the file and restarted. Same result. This file is also not directly downloadable that I could find. Apparently the IDE is not much use without it, as when dismissing the message and starting an new project I get:

_No boards.txt files found in the arduino hardware folders. I looked in:
/home/user/Arduino/hardware
/home/user/sloeber/arduinoPlugin_

And I can't create a sketch.

Tried numerous times on two separate PC's. Linux 32 and 64 bit.

jantje, 2017-09-12:
Have you looked at the video's? Did you install the product?
Which version?
Did you get an error?
Do you have spaces in your paths?

ouryperd, 2017-09-24:
Have you looked at the video's? Did you install the product?
Yes and yes.
Which version?
4.1
Did you get an error?
Yes
Do you have spaces in your paths?
No

I have used Eclipse for the last 12 years for Java, Ruby, Perl, Python and C development. I know Eclipse's workings quite well. Never had an issue like this before.

jantje, 2017-09-24:
Strange
Note that Sloeber is not a Standard eclipse product, it is based on cdt which is on its own already "special" and sloeber adds  "arduino ide" functionality which doesn't match eclipse ideology as well.
What I'm trying to say is that maybe your 12 years experience with eclipse is not helping you out here.
If you can not create a sketch:
Try to install a platform. in menu->arduino->preferences->arduino->boards
and try again to create a sketch.
The examples are not  needed to run sloeber and as the error says they are available.

rlogiacco, 2017-09-24:
Maybe this part of the message might be helpful

> Can't extract file/home/user/sloeber/arduinoPlugin/examples/01.Basics/AnalogReadSerial/AnalogReadSerial.ino, file already exists!

I believe Sloeber has already partially unpacked the ZIP file until it found it truncated and aborted. 
You can try removing the `/home/user/sloeber/arduinoPlugin/examples` folder and try again, or simply try to delete the whole product folder `/home/user/sloeber` and start again. 

I believe the original issue _Failed to extract tar.bz2./home/user/sloeber/arduinoPlugin/downloads/examples_Arduino_1_6_7.zip
Truncated ZIP file_ was caused by some network issues.
We might want to make this phase of the install process a little more robust: what do you thing @jantje ?

jantje, 2017-09-24:
Implement a checksum for download you mean?

ouryperd, 2017-09-25:
I found the problem. My internet proxy/firewall server (IPCop) cached a partially downloaded version of examples_Arduino_1_6_7.zip

This meant that even if I deleted the file and restarted Sloeber, it kept downloading the broken file. I configured IPCop not to cache files from the remote address, deleted the zip as well as unpacked examples, and voila - it worked.

Thanks. Hope this helps in some way. #818 can be closed.

closed 2017-09-25 by ouryperd

jantje, 2017-09-25:
Network problems. The problem of the future.
I'm happy it is fixed.