Sloeber/arduino-eclipse-plugin #819: Debug esp32 FreeRtos applications

freeck, 2017-09-13:
I cannot imagine being the first, but I could not find any procedure on the internet for enabling source level debugging with an JTAG-ICE, using Eclipse and the Sloeber-plugin.
I finally succeeded, so if there is any interest I could publish the procedure.

jantje, 2017-09-13:
yes there is :-)))

jantje, 2017-10-08:
FYI
We have been working hard to get regression tests in the build.
Now this is done I want to add my hardware repositories to the regression test as some changes in Sloeber make them fail.
So yes I'm still interested but things do not progress as fast as I would like.

freeck, 2017-10-08:
No problem, in the mean time I am able to further review and improve the documentation on the installation and debugging process.

jantje, 2017-12-11:
Have you seen these video's on debug on esp32 and zero?
https://youtu.be/WFzVzgEA8Fo
https://youtu.be/DXTo4-0pWKA
I think I will make time for doing the same with esp32 beginning next year.
Would that fit your agenda?
How is the doc going?
If you prefer dutch we can do the discussion on the Arduino forum like we did for esp8266

freeck, 2017-12-12:
Yes I have seen your video's. Concerning the esp32: Perhaps you have seen this video https://m.youtube.com/watch?v=xJyNLcoG7Q0 ? It is in German, but the video is very instructive.

As I did not receive any reactions (accept from you) on my plans to make a video, the priority has dropped (I am still very curious about how people can develop software especially for embedded systems, without a proper debugger).

OK, I will make a document for "setting up the debugging environment for a esp32-target" before the first of January..

freeck, 2017-12-12:
...2018

jantje, 2017-12-12:
I created a thread in the dutch part of the Arduino forum
http://forum.arduino.cc/index.php?topic=516773.0

Develectronic, 2018-06-18:
dear freeck,
I am very very interested in debugging the esp32 in slober. I looking for a tutorial on the internet for hours, but there is absolutely nothing. And when I got that right, you have found a way to to that. I would be so thankfull, if you could give me the instructions how to configure sloeber to debug the esp32.

Thanks a lot.

Develectronic, 2018-06-18:
As you also mentioned above, I cannot understand how other people developing software without debugging. 
But I guess most peoble do not know that it is even possible and others have just no idea how to get it working (like me). 
That would be a second christmas if you produced a document by the end of this week.
Thanks a lot!!!!!

Develectronic, 2018-06-18:
and yes I know there is jtag interface required. I've alread an xds100v2 or an olimex arm-usb-ocr ( https://www.olimex.com/Products/ARM/JTAG/ARM-USB-OCD/ ) when neccessary I buy another...

freeck, 2018-06-21:
Hi there, Am I saved by the bell even before the weekend? :)
Have you already read this document? 
https://esp-idf.readthedocs.io/en/latest/api-guides/jtag-debugging/
It does not describe the integration with Sloeber but the most difficult part i.e. the installation and testing of the jtag-interface with the esp32 target and usage of openocd with the gdb for the extensa-processor. 
So could you first try this?

Develectronic, 2018-06-21:
Hi, 
yes I've seen this side already, but I thougt that it works only for IDF project, when it is more or less the same for sloeber I can try it and may ask some specific questions. 
But is there an relation between this FreeRTOS and the JTAG debugguing? I should be able to debug every "arduino-sketch", right?

freeck, 2018-06-21:
1: JAG/openocd is meant as a standard interface with any IDE or via a command line interface.
So first try to install and test openocd and test it via a command line as described in the document.

2: "debug every arduino-sketch"? it depends whether the processor you use has a jag-interface... the Atmega328 does not have a Jtag-interface. FreeRtos is used on the esp32 standard environment. It had no link with jtag as such.

jantje, 2018-06-21:
If you install the zero package you have open ocd in sloeber
If the communicatio, is openocd the "sloeber" side is the same as the zero
https://www.youtube.com/edit?o=U&video_id=DXTo4-0pWKA

PS I say "sloeber" as it is actually gnumcu plugin that delivers this debug stuff

freeck, 2018-06-21:
I was not aware that this video I made was still in the cloud ! 😂
I now feel myself more or less obliged to finish the video....

Develectronic, 2018-06-21:
Ok, thank you for the tips. The video is no longer accessible (or link does not work), but can I use this one for the zero package installation? https://www.youtube.com/watch?v=WFzVzgEA8Fo 
Then I need to do the esp32- and JTAG hardware specific configuration?

jantje, 2018-06-21:
the link was wrong this the correct one https://youtu.be/DXTo4-0pWKA

freeck, 2018-06-21:
Great video!  As it bases on an ARM-processor I will add the esp32 specific items. 
So @Develectronic be patient for a day or two....

freeck, 2018-06-22:
Hi folks, I produced a document I placed on Onedrive. A concept so any comment is welcome.
Are you able to open it?
https://1drv.ms/w/s!AgjNziQJTfNOhrlUJad5txK8jXhrTA

jantje, 2018-06-24:
I can open the files but I won't be able to address this in the first coming days.
I did however create a json to install esp32 easily. I only tested on windows so please feedback if it works or doesn't work for you
https://raw.githubusercontent.com/jantje/arduino-esp32/master/package/package_Espressif_esp32_index.json

freeck, 2018-06-25:
It works OK with Sloeber4.1, I with test it with Sloeber4.2 asap..

freeck, 2018-07-05:
Jantje,
I have reached another milestone: after a lot of trial and error, programming the flash via the jtag-interface is now working. I have updated the installation tutorial:
https://1drv.ms/w/s!AgjNziQJTfNOhrlUJad5txK8jXhrTA 

Two minor issues remain:
- it would be nice to link the download button to the jtag-flashing procedure 
- in the debug configuration the working directory macro/variable does not work (at least in my environment). I solved this by including the full path of the bin-file. I lack the Eclipse-knowledge to solve this....

freeck, 2018-07-06:
And a video on YouTube: https://youtu.be/hxOB7Qp0yM0

jantje, 2018-07-11:
Sorry for the silence. I'm really busy right now and will be until September. But I do hope to get this to work for the fri3d camp in august.

jantje, 2018-07-21:
Great video :-) I added it to my sloeber channel.
Next is reading the installation tutorial

jantje, 2018-07-21:
Reading the doc I wonder if there is any need for a change in Sloeber :-D
In the doc I read "Only two hardware breakpoints are supported."
I don't really understand the user impact of this. I guess there are hardware breakpoints and software breakpoints but what are the differences? Do you know by now?

freeck, 2018-07-21:
See also: http://esp-idf.readthedocs.io/en/latest/api-guides/jtag-debugging/tips-and-quirks.html

In case the user defines more breakpoints you have to add the IRAM-directive in the header of the function(s) you want to debug. In case you dont you will get an error message in de the debug-console of gdb. It has no consequences for Sloeber...

jantje, 2018-07-21:
That is exactly what I wanted to know :-)
Thanks

easyvolts, 2018-07-30:
Hi freeck,

thank you for the manual, it's very helpful. I used it some days ago and had a problem with esp32 jtag not detected by the openocd, the problem was in esp32 board. It needs to have JTAG debugging enabled because one of JTAG pins is used by bootloader (this is strange enough because I thought that openocd connects to the device under reset, but it worked for me). JTAG can be enabled in menuconfig. Probably it would be good to add this information to the document.

freeck, 2018-07-30:
@easyvolts Thanks for your comment. 
The instruction was specifically written for Sloeber/Windows…. I do not know whether menuconfig can be executed within Sloeber or as a stand alone program?
Perhaps someone can tell me, then I have learned something today! :-)

freeck, 2018-08-23:
Great, You have passed the most "difficult part" of the installation process i.e. getting the jtag-driver work together with gdb. 
The next step is to follow the instructions decribed in https://onedrive.live.com/view.aspx?resid=4EF34D0924CECD08!105684&ithint=file%2cdocx&app=Word&authkey=!ACWnebcSvI14a0w .....

RA5040, 2018-08-23:
Many thanks Fred (I've just seen your first name on your article ... hope you don't mind the familiarity :)).  Since posting I have installed gdb/jtag on Eclipse (following the Espressif instructions) and successfully built and debugged the blink example. 

But not without some problems of course.  The main problem was that Eclipse (with the Espressif settings) does not start OpenOCD. I get an error saying "Failed to execute MI command:
-target-select remote localhost:3333".  As far as I can see there is nowhere to set up the debugger information required (the command line to start it is bin/openocd -s share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp32-wrover.cfg. In Eclipse I can't see where to set up the interface and board configs .... but perhaps after reading your article all will be clear!).

I've written up the steps here: http://www.irelandupclose.com/customer/esp32/jtag/JTAG_ECLIPSE.html

I must say that it's quite a buzz seeing the debug environment pop up!  It was great on the command line, but on Eclipse ... double-great!  

Now for Sloeber ... fingers crossed ...

freeck, 2018-08-23:
Did you compare your setup with the screendumps  I added in my instruction? From what I read in your instruction you havent....

RA5040, 2018-08-23:
Yes, I'm going to do that ... I just wanted to build up to it from the Espressif instructions (command line, then Eclipse ... and once I had that working to move on to Sloeber).

The thing is .... which is crazy ... I need to install V4.3 of Sloeber and I can't find it anywhere.  The link to the plug-in: http://eclipse.baeyens.it/update/V4/stable is broken.

Can you ... or Jantje ... tell me where to get it?

RA5040, 2018-08-23:
OK ... I can see that the plugin comes with the latest version of Eclipse. But I can't get it to run so I need some help from jantje.

RA5040, 2018-08-26:
Hi Fred,
I've tried your instructions and so far no luck.

Here are the settings and the error log.  As you can see it is failing to find "target/esp32.cfg" in the board config (esp32-wrover.cfg).  
[image]

The manual command (which works, is this:)
bin/openocd -s share/openocd/scripts -f interface/ftdi/esp32_devkitj_v1.cfg -f board/esp32-wrover.cfg

so I tried this line in the Config options:
-s C:\msys32\home\Robert\esp\openocd-esp32\share\openocd\scripts -f C:\msys32\home\Robert\esp\openocd-esp32\share\openocd\scripts\interface\ftdi\esp32_devkitj_v1.cfg -f  C:\msys32\home\Robert\esp\openocd-esp32\share\openocd\scripts\board\esp32-wrover.cfg

This time it didn't fail on the esp32.cfg (so I guess it found the file), but it did fail with this error message:

Error in final launch sequence
Failed to execute MI command:
continue
Error message from debugger back end:
Warning:\nCannot insert breakpoint 1.\nCannot access memory at address 0x400d0ad8\nCannot insert breakpoint 3.\nCannot access memory at address 0x400d0ae2\n
Failed to execute MI command:
continue
Error message from debugger back end:
Warning:\nCannot insert breakpoint 1.\nCannot access memory at address 0x400d0ad8\nCannot insert breakpoint 3.\nCannot access memory at address 0x400d0ae2\n
Warning:\nCannot insert breakpoint 1.\nCannot access memory at address 0x400d0ad8\nCannot insert breakpoint 3.\nCannot access memory at address 0x400d0ae2\n

Any ideas?

BTW, I did get the debugger to work well with C++ programs, but it's a different setup and it didn't work with Sloeber.

Cheers

Robert

freeck, 2018-08-26:
Hi Robert,

Your setup seems to be OK.
It seems as if you defined too many breakpoints; so I would suggest to delete all the breakpoints and restart the debug process and see what happens..

freeck, 2018-08-27:
Aha...so somewhere the debugger tries to insert a breakpoint also in case there seems to be no breakpoints defined...
Question: At the location of xtensa-elf-gdb.exe: is there a file called gdbcmds? If so the content of this file could interfere with the startup configuration (there could be a breakpoint set at a non-existing location.... If so, delete this file....

RA5040, 2018-08-27:
No, unfortunately there's no gdbcmds file, just the xtensa exes and one config file: xtensa-esp32-elf-ct-ng.config (or anywhere else in the msys32 folders).

I tried to debug in RAM but I get the same error, just at a different address. 

So even with no breakpoints at all and no halt at loop, there is still this error:

Warning:\nCannot insert breakpoint 1.\nCannot access memory at address 0x400e8c28\n

Regarding the breakpoint: is there a breakpoint automatically set at main?  That would explain the breakpoint issue.

Which leaves 'Cannot access memory at address xxxx'.  I had thought that this was in reserved memory, but it isn't. So why the error?

When you say you could reproduce the error with gdbcommands, did you also get the memory error?

freeck, 2018-08-27:
I have a problem... I am not able to reproduce the error message anymore. It now only occurs when I define more than 2 breakpoints. Then I get the same error message as you except at another address.

> Error in final launch sequence
Failed to execute MI command:
continue
Error message from debugger back end:
Warning:\nCannot insert breakpoint 3.\nCannot access memory at address 0x40080e24\n
Failed to execute MI command:
continue
Error message from debugger back end:
Warning:\nCannot insert breakpoint 3.\nCannot access memory at address 0x40080e24\n
Warning:\nCannot insert breakpoint 3.\nCannot access memory at address 0x40080e24\n

A few things I can imagine:
1: I have copied "xtensa-esp32-elf-gdb.exe" to a separate directory (no interference with some configuration file)
2: Which version of "xtensa-esp32-elf-gdb.exe" do you use? (I have tried 2 versions: September 2017 and June 2018, both work).

RA5040, 2018-08-27:
The toolchain is the 20180110 version and the openocd is 20180724 ... or at least these are the files that I downloaded (pretty sure I installed them). But I'll get the latest stable releases and have another go. And I will also try moving the xtensa...gdb to a different foler.

At least it's (hopefully) a good thing that you are getting a similar error: but why my system is trying to insert breakpoint 1 is certainly a puzzle!  I don't suppose Jantje would have any ideas?

Another thing I'll do is to try it on a different sketch and also on Sloeber 4.3 (I running 4.2 at the moment). Which version are you using?

freeck, 2018-08-27:
I am running on Sloeber 4.1 . Tomorrow I will try both 4.2 and 4.3.
According to this link https://github.com/espressif/openocd-esp32/issues/51 
Openocd.exe relates to the error I observed , but for me it did not work.

RA5040, 2018-08-27:
Do you mean the <2 breakpoints? I also saw that somewhere else (but as a programmed restriction). Not sure where that was though. But if you can do more than 3 bpts then that's great.
I'm going to do the following:
i. Try on a C++ program in case it's a Sloeber issue (I couldn't run GDB Hardware Debugging on Sloeber either, but it works on C++).
ii. Move the xtensa-gdb.exe somewhere else
iii. Make absolutely sure the toolchain and OpenOCD are up to date (pretty sure they are)
iv. Try on Sloeber 4.3
v. Try GDB Hardware Debugging again to see if it's the same issue (I can't remember what the error was)

If you have any other suggestions let me know. BTW ... thank you very much for the help!! I would be lost on my own.

freeck, 2018-08-27:
No, I was able to set 2 breakpoints. One should be able to set more software breakpoints using the IRAM directive, but it does not work for me.

RA5040, 2018-08-27:
OK ... well for now I would be happy to be able to set 1 breakpoint!

RA5040, 2018-08-28:
https://github.com/espressif/openocd-esp32/issues/51 ... seems the 2-bpt issue is fixed

freeck, 2018-08-28:
Yes but it did not work with my configuration. I wil double check my configuration and test it once again. Are you sure that things works OK for you in case using command lines? Could you log the gdb-output and compare it with the log using Sloeber?

RA5040, 2018-08-28:
I've run a c-program test with 4 breakpoints using OpenOCD, no problem.  I ran the same program with Hardware debug with 6 breakpoints, also fine.

BTW ... one issue I had was that in my first test I was just assigning some values to a couple of variables and the debugger went mad after first Next with lots of:
Info : Target halted. PRO_CPU: PC=0x400D1F49 (active)    APP_CPU: PC=0x400D17AE 
Info : Target halted. PRO_CPU: PC=0x400D1F49 (active)    APP_CPU: PC=0x400D17AE 
...

I saw that the compiler was optimising out the code because the variables were not used, so I set them to volatile and then everything worked OK.  So that's something to watch out for (the debugger doesn't really like empty programs!).

RA5040, 2018-08-28:
Going back to OpenOCD on Arduino:
**i. Try on a C++ program in case it's a Sloeber issue (I couldn't run GDB Hardware Debugging on Sloeber either, but it works on C++).** WORKS OK
**ii. Move the xtensa-gdb.exe somewhere else**  NO DIFFERENCE
iii. Make absolutely sure the toolchain and OpenOCD are up to date (pretty sure they are): NOT DONE YET
iv. Try on Sloeber 4.3: NOT DONE YET
**v. Try GDB Hardware Debugging again to see if it's the same issue (I can't remember what the error was)**  DOESN'T WORK ... SIMILAR ISSUE (see below):
Error: esp32: Faied to read insn (-4)!
Error: esp32: Faied to add SW BP!
Error: can't add breakpoint: resource not available

So, to summarize so far: [Hardware: ESP32 Wrover Kit; Windows 10; Propably latest versions of tooltchain, OpenOCD, ESP-IDF (to be confirmed)]:
-  Both OpenOCD and Hardware debug work on C programs on Eclipse (that's something!!); 
- Neither OpenOCD nor Hardware debug work on Arduino programs on Eclipse Sloeber V4.2.

TO BE CONTINUED :)

freeck, 2018-08-28:
I have tried 4.3. I observe that the gdb-client has minor GIU changes. Good news is that the debuggers works fine. Bad news: I still cannot insert more than 2 breakpoints.
error log:
Warn : xtensa_add_breakpoint: max HW slot reached, slot=2
Error: No flash at address 0x400d08d0
Error: Failed to get flash bank (-4)!
Error: esp32: Failed to add SW flash BP!
Error: can't add breakpoint: resource not available

RA5040, 2018-08-28:
Good and bad for you ... but not good for me: I get the same error.  So at least this has nothing to do with the version of Sloeber. But it would seem to have something to do with Sloeber/ESP32 Dev Module.

In fact, I'm thinking that perhaps that is the problem, because I get the identical error if I change to the ESP Wrover module (which doesn't have JTAG).  What do you think?  Could that be it?

freeck, 2018-08-28:
I dont understand what you mean by "perhaps that is the problem, because I get the identical error if I change to the ESP Wrover module (which doesn't have JTAG)."  As you dont have a jtag, what error are you refering to?
 I will first test the command line interface trying to define more breakpoints in order Sloeber is excluded.

freeck, 2018-08-28:
Well I have tested gdb and openocd without Sloeber, and get the same error messages. I cannot insert more than 2 breakpoints. 
So we may for now exclude Sloeber as the source of the problem....

RA5040, 2018-08-28:
We're talking about two problems at once:
- multiple breakpoints: I don't have a problem with this because I can set multiple breakpoints both with OpenOCD and Hardware gdb.  However you do have a problem because you can only set 2 breakpoints.  I didn't suggest that this was an issue with Sloeber, although as I can't run gdb with Sloeber I can't verify it (but you seem to be excluded Sloeber, so that's good).
- the second problem, which is the one I have, is that I cannot run gdb (either OpenOCD or Hardware gdb) with Sloeber: I get the "Cannot set breakpoint error" and the debug crashes out. But I can run both OpenOCD and Hardware debug with C and C++ apps.  The ESP Wrover Kit does have built-in JTAG and should work with Sloeber and OpenOCD (or Hardware debug), but it doesn't. Trying to debug on it behaves exactly as it does if I select the ESP Wrover module (which does not have JTAG built-in).  So what I was wondering is this: is it possible that Sloeber is incorrectly selecting the ESP Wrover Kit, so that I am effectively trying to run gdb on a board that does not support it?

jantje, 2018-08-28:
>the second problem, which is the one I have, is that I cannot run gdb (either OpenOCD or Hardware gdb) with Sloeber

have you run gdb from the command line? maybe you are missing a library.

freeck, 2018-08-28:
I have found the solution, at least for my problem, but this could be a hint for you as well.
I have downloaded the latest version of esp32.cfg , from the sources of OpenOcd! Not the latest zip-file. 
This file seems to "control" the behaviour of gdb and openocd. Now I am able to set multiple breakpoints. Hope this might help you as well.

Answering to your question: Did you you install 2 platforms? If so delete one and test once again.
But I would bett on the configuration file of the target!

RA5040, 2018-08-28:
Jantje ... no I haven't tried to run gdb from the command line on Sloeber code (but I have on C/C++ code). I'll try that.
Fred ... that sound very promising (and I'm delighted to hear you've fixed your problem! ...GREAT!!).  I will need to use the ESP Wrover because it says that:
#   For ESP32-WROVER module use 'esp32-wrover.cfg' configuration file
#   ESP-WROOM-32 and ESP32-WROVER have different flash voltage setting
So I'll download this one and try it:
https://github.com/espressif/openocd-esp32/blob/master/tcl/board/esp32-wrover.cfg
Mind you ... this is one year old so it's probably the same one I'm using. Fingers crossed!

RA5040, 2018-08-28:
Is this the esp32.cfg you used?
https://github.com/espressif/openocd-esp32/blob/master/tcl/target/esp32.cfg

freeck, 2018-08-28:
Oh....I thought you had tested command line debugging already....that will reveal the problem! (I think l:)). 

Concerning 1 year old cfg file: No problem, it includes the latest version of esp32.cfg

RA5040, 2018-08-28:
I downloaded the esp32.cfg from the link above and it is identical to the one I already have.  When you say you downloaded it from the the sources of OpenOCD, could you give me the link please? I can't find it.

I've run the sketch from the command line and it's certainly a lot better. The program halts at loop and I can do a Next. However it crashes on the second Next:

Temporary breakpoint 1, loop () at ..\Blink.ino:24
24      void loop() {
(gdb) l
19        // initialize digital pin 13 as an output.
20        pinMode(2, OUTPUT);
21      }
22
23      // the loop function runs over and over again forever
24      void loop() {
25        digitalWrite(2, HIGH);   // turn the LED on (HIGH is the voltage level)
26        delay(1000);              // wait for a second
27        digitalWrite(2, LOW);    // turn the LED off by making the voltage LOW
28        delay(1000);              // wait for a second
(gdb) n
25        digitalWrite(2, HIGH);   // turn the LED on (HIGH is the voltage level)
(gdb) n
Warning:
Cannot insert breakpoint 0.
Cannot access memory at address 0x400d0ae2

__digitalWrite (pin=0 '\000', val=0 '\000')
    at C:\Users\Robert\git\arduino-esp32\cores\esp32\esp32-hal-gpio.c:173
173     {
(gdb)

It appears to be unable to insert a breakpoint.  Does this mean something to you?  BTW, I set no breakpoints in the sketch.

[54 more replies]

closed 2020-01-06 by freeck