Sloeber/arduino-eclipse-plugin #820: Serial Plotter export bugs

mlasevich, 2017-09-14:
On export of plotter data, the generated "CSV" file seems to have two issues

Issue 1 - There is no newline between the header row and the first data row

Issue 2 - In the header row this is fine, but in the data rows the code attempts to write a ';' as a separator character, but instead writes digits "59" (ascii value of ';')

This is happening on OSX on Sloeber version 4.1.0.201705252335

Example output:

      Channel0;Channel1;Channel2;Channel3;Channel4;Channel5;595959595959
      595959595959
      595959595959
      595959595959
      595959595959
      595959595959
      595959595959
      595959595959
      595959595959
      595959595959
      ...

mlasevich, 2017-09-14:
For what its worth, upon looking at code, it appears the bugs are actually in the [getData() method of org.eclipse.nebula.widgets.oscilloscope.multichannel.Plotter](https://github.com/eclipse/nebula/blob/master/widgets/oscilloscope/org.eclipse.nebula.widgets.oscilloscope/src/org/eclipse/nebula/widgets/oscilloscope/multichannel/Plotter.java#L464-L478) class 

I will file a bug there, but it can be easy to override it in Sloeber code too

jantje, 2017-09-14:
I see why the newline is missing but why is it 59 instead of ';' ?

mlasevich, 2017-09-14:
I suspect it is a combination of order of precedence (+ is evaluated before +=) combined with using a char instead of a string for delimiter (single quotes vs double quotes). Because the line is `String += int + char;`, it probably adds the int to char first, then add the result, as integer, to the string.

Given that, my bug description is slightly off, as instead of using 59, it is actually using (value+59) and no delimiter - just that value is often 0

jantje, 2017-09-15:
thanks, makes sense.

mlasevich, 2017-09-18:
FWIW, the patch submitted to Nebula was accepted/merged: https://bugs.eclipse.org/bugs/show_bug.cgi?id=522323 - (not sure what this means for when it can be just used in Sloeber)

jantje, 2017-09-18:
Basically as soon as you do a update in eclipse it should be fixed.
The next nightly release should contain it as well.
The reference to nebula is currently the latest (not smart but we needed it at some time and nothing bad happened since then)

mlasevich, 2017-09-19:
Confirmed! Ran update, it updated to latest Nebula Oscope, and voila - the data export works like a charm :-)

closed 2017-09-19 by mlasevich