Sloeber/arduino-eclipse-plugin #822: Adding Libraries leads to Errors

Seppeltronics, 2017-09-15:
I just updated to 4.1 which broke sloeber. When adding a linked folder, e.g. for a library or changing the path, sloeber goes crazy with errors and claims it can not link files to folders: 

Error-Message:

Could not create folderlibraries/Adafruit_ILI9341
Cannot create linked resource '/sloeberDisplayTouchTest/libraries/Adafruit_ILI9341'.  Files cannot be linked to folders.

jantje, 2017-09-16:
Please provide a step by step way to reproduce this.
Which os are you on?
Are bthere stack traces in the error log?

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

Gord1, 2017-09-19:
I'm having issues with libraries too.  Using Linux Mint with today's download and updated.  It expanded itself in /home/gordon/Downloads/sloeber/
I don't see how to set up a private library.  I followed you tutorial http://eclipse.baeyens.it/how_to.shtml#/j
and got down to "Create a subfolder with the library name in you private library folder" but was unsure where that should be.  My workspace is the default /home/user/Documents/sloeber-workspace/.
I've tried creating library folders under both the workspace and project.  When I use Arduino>  Add library to selected project I get a box "Import Arduino Libraries" like you get *but* it only shows the libraries that have been imported the library manager Arduino>preferences>Library Manager.  There isn't any Hardware provided Libraries, Arduino Libraries, Personal Libraries.

AH... OK, I found where the private library setting under Ardunio>Preferences> Arduino .  That was set to a non-existent folder and the same for  Private Hardware Path.

OK now setting the private library path to a folder in my workspace and putting the .h and .cpp files in a folder under it, I can now see them in Arduino>Add Library ...  That works now and I can compile.

But this just shows all the downloaded libraries mixed in with my private libraries.

How to I separate them into groups like you have in your tutorial?  e.g. Hardware, Arduino and Personal.  The downloaded libraries default to /home/gordon/Downloads/sloeber/arduinoPlugin/libraries  which are kind of hidden.

jantje, 2017-09-19:
@Gord1 
>How to I separate them into groups like you have in your tutorial? 

I don't exactly recall why but I had to remove this separation.
It looks to me like all other issues are fixes.

jantje, 2017-09-25:
is there still a issue here or can this be closed?

Gord1, 2017-09-25:
Sorry, Yes it can be closed (at least for my part).  But I suggest 2 things to make it easier for newbees like me following your tutorial http://eclipse.baeyens.it/how_to.shtml#/j.

1) Make a note where to find the settings for the path to the private library. e.g. Ardunio>Preferences> Arduino.  It took me a long while to find that missing bit of info.

2) You could update the screen shot of "Select the Arduino Libraries" to show the libraries as they are instead of separated into the different groupings.  It gave me the impression that I was doing something wrong as mine weren't grouped like yours.

Thanks,
Gordon

jantje, 2017-09-25:
The website needs some tlc I know.
If you got the skills I'd be pleased with some help. The website is part of this repository.

jantje, 2017-11-21:
Updated

closed 2017-11-21 by jantje