Sloeber/arduino-eclipse-plugin #823: Downloading wrong package_index.json files

jonathanmueller, 2017-09-17:
When using a "third party index url", it seems that before downloading from the specified url, you first check if a copy of the file is available at `http://eclipse.baeyens.it/download/<filename>.json`.

This makes it impossible to download use newer versions of packages with the same filename.

For example, adding the url for the prerelease version of the esp8266 package
`https://github.com/esp8266/Arduino/releases/download/2.4.0-rc1/package_esp8266com_index.json`
to the third party index url's results in downloading the file from
`http://eclipse.baeyens.it/download/package_esp8266com_index.json`, which is the current stable release (2.3.0), making it impossible to use the new version.

jantje, 2017-09-17:
It is the esp8266 package that made me do this. See #743

jonathanmueller, 2017-09-17:
I see... Thanks for answering :)

Would it be possible to check for the alternative download by including the complete url in the request to baeyens.it in order to avoid future file name collisions (though I assume this is a very unique case)?

If that would be too much sensitive information being sent to your server, maybe a hash of a normalized form of the url?

jantje, 2017-09-17:
I was actually hoping this issue would be fixed one day and could be removed.
So I don't fell like adding lots of "special tricks".

jonathanmueller, 2017-09-17:
You're probably right...
Looking through #743 I discovered the "Update local json files" option which I had enabled when playing around with the plugin, overwriting the json file at every startup.
I guess the best solution still is disabling that option and manually downloading the file to the arduinoPlugins folder.

jantje, 2017-09-17:
>Looking through #743 I discovered the "Update local json files" option which I had enabled when playing around with the plugin, overwriting the json file at every startup.

That is what this option does indeed.

>I guess the best solution still is disabling that option and manually downloading the file to the arduinoPlugins folder.

If you want/need advanced control over your development environment I think this is the way to go.

closed 2017-09-20 by jonathanmueller