Sloeber/arduino-eclipse-plugin #825: Check for cdt indexer options

jantje, 2017-09-18:
With more and more functions relying on a "good working" of the indexer it is a good idea to check whether the indexer settings are "as expected by sloeber".
My settings are:
[image]
note that index source files not included in the build must be turned off.
People using multiple configurations should use the active configuration to build the indexer.
I don't think other settings are really important.
Well indexer on is, should test what Sloeber does when the indexer is off.

jantje, 2017-09-18:
apparently index unused headers must be turned off as well.

jantje, 2017-11-04:
I added checks for these values directly reading the instancescope.
The user needs to change the value in windows->preferences->c/c++->indexer
This warning can not be disabled

closed 2017-11-10 by jantje

climategadgets, 2020-04-13:
This behavior breaks at least ESP-IDF projects.

Changing IDE configuration, especially without a warning, especially from default settings, is... time consuming for the unaware.

It would be a welcome feature if the installer asked nicely to change this setting, and warned the user about consequences of not changing it.

climategadgets, 2020-04-13:
This behavior breaks at least ESP-IDF projects.

Changing IDE configuration, especially without a warning, especially from default settings, and especially if Sloeber installation failed because of transient errors is... time consuming for the unaware.

It would be a welcome feature if the installer asked nicely to change this setting, and warned the user about consequences of not changing it. And only after making sure that all the prerequisites were successfully obtained.

jantje, 2020-04-13:
@climategadgets Not sure what you are talking about.
The indexer does not influence the building of the project.
If the indexer includes unwanted libraries you can disable this feature in preferences->arduino.

As to  
>It would be a welcome feature if the installer asked nicely to change this setting, 

1) This issue is about sloeber checking these values and telling you "nicely" at startup when Sloeber thinks these are set wrongly.
2) I think Sloeber has no lessons to learn from you in regards to "asking nicely"

climategadgets, 2020-04-13:
The installer immediately changes the indexer settings without giving any indication that it is going to do so. The side effect of that is Eclipse immediately loses track of references, all open files get red marks, and code cannot be navigated anymore.

uzi18, 2020-04-13:
@jantje have same situation in time of editing file indexer marks all with red and sometimes recover sometimes not, so code completion works unreadable.
Give me some hints how to investigate this?

jantje, 2020-04-13:
@uzi18  First is always: rebuild the index.