Sloeber/arduino-eclipse-plugin #826: Unable to upload code to Adafruit Trinket

fovea1959, 2017-09-19:
Fresh Install of Sloeber 4.1 on Ubuntu 16.04 64-bit. Added the board library for Adafruit `https://adafruit.github.io/arduino-board-index/package_adafruit_index.json`, put together simple project for Adafruit Trinket, use the Adafruit AVR platform folder v1.4.9, select correct board type, default upload method. There are no values available in the "Port" pulldown (since Trinket do not present to OS as a serial port). There is 

Project builds, but upload fails. The upload console says

```
Starting upload
using arduino loader
Com port not found. Skipping reset
failed to upload
```

and the eclipse error logs has this traceback

```
java.lang.NullPointerException
	at io.sloeber.core.tools.uploaders.arduinoUploader.uploadUsingPreferences(arduinoUploader.java:75)
	at io.sloeber.core.tools.uploaders.UploadSketchWrapper$UploadJobWrapper.run(UploadSketchWrapper.java:177)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)
```

I am able to use Arduino IDE 1.8.4 fine on this computer and Trinket, so permissions and all the OS level stuff appear to be good.

Unsure how to fix.

fovea1959, 2017-09-19:
...also, when uploading using Arduino IDE, Trinkets are programmed using USBTinyISP. Sloeber does not give me the option to select that upload method when a Trinket board is selected. Is the real issue that the upload method pulldown is not filled in with the correct values? Can the pulldown be filled in with all known values (like what appears if I select a Leonardo?)

jantje, 2017-09-21:
As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

There have been quite some changes in regards to upload in the nightly so this may work in the nightly.
It might be that the programmer appear in the nightly (Not sure I released the change yet though)
So lets first try the nightly and go from there.

jantje, 2017-09-25:
do you have a update?

fovea1959, 2017-09-25:
Update: Using io.sloeber.product-4.1.0-20170911.231844-88-linux.gtk.x86_64.tar.gz:

Upload now does this:

```
Starting upload
using arduino loader
Com port not found. Skipping reset

Launching/home/wegscd/bin/sloeber//arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -C/home/wegscd/bin/sloeber//arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/etc/avrdude.conf -pattiny85 -cstk500v1 -P -b19200 -D -Uflash:w:/home/wegscd/Documents/sloeber-workspace/Tedst/Release/Tedst.hex:i 
Output:
avrdude: ser_open(): can't open device "-b19200": No such file or directory

avrdude done.  Thank you.

avrdude finished
upload done
```

jantje, 2017-09-25:
did you set the com port?
could you select USBTinyISP

fovea1959, 2017-09-26:
There were no pulldown choices available for port. The only choice for "Upload Protocol" was 'Default'.

I suspect that if there was no port selected, the -p was left off the avrdude download command, we'd have a good chance of this working...

jantje, 2017-09-26:
>I suspect that if there was no port selected, the -p was left off the avrdude download command, we'd have a good chance of this working...

maybe true  but this is not how it is supposed to work.

can you try to copy the arduino avr programmers.txt to your platform?

fovea1959, 2017-09-26:
I am happy to; there appear to be copies in my arduino IDE installation. 

Where do I copy them to?

jantje, 2017-09-27:
[sloeberinstall]/arduinoPlugin/packages/adafruit/....
next to the boards.txt containing Adafruit Trinket

fovea1959, 2017-09-27:
Bingo!

Fresh install of sloeber product daily, Linux 64 bit, add the adafruit board library. shutdown sloeber.

`cp aa/arduino-1.8.4/hardware/arduino/avr/programmers.txt ~/sloeber/arduinoPlugin/packages/adafruit/hardware/avr/1.4.9/`

Bring sloeber back up, and it built and downloaded flawlessly!

jantje, 2017-09-28:
thanks for the feedback. I need to think about how to handle the programers.txt of arduino

fovea1959, 2017-09-28:
Thank *you*. Let me know when you want things tested; I also do ESP8266, so I'm a decent test dummy...

fovea1959, 2017-09-28:
It appears that the Arduino IDE "falls back" to the arduino programmer definitions if there are none included with a third party library. That seems to be a reasonable behaviour, though tricky since sloeber appears to support having multiple versions of libraries available?
 

    On Wednesday, September 27, 2017 9:28 PM, Doug Wegscheid <dwegscheid@sbcglobal.net> wrote:
 

 Thank *you*. Let me know when you want things tested; I also do ESP8266, so I'm a decent test dummy...

jantje, 2017-09-28:
>though tricky since sloeber appears to support having multiple versions of libraries available?

If you mean multiple versions of a platform, then indeed.
I changed sloeber to include "referenced platforms". Apparently this board does not reference Arduino.
Note that finding the referenced platform in a multi platform version environment leads to making choises. In the nightly you can select between the arduino platform selection way (= the latest version) or not (=the json referenced version and if no version in the json the latest)
Maybe a help button with programmers.txt copy instructions may be a good thing.

fovea1959, 2017-09-29:
hokay. let me know when we have a nightly with the changes in it; I'll test it!

jantje, 2017-10-20:
@fovea1959 
I implemented following logic. If no programmers.txt is found in the referencing and upload referenced platform use the platform?txt file or Arduino for this architecture.
This should be available from the nightly in about 10 hours from now.
Can you test this?

closed 2017-11-03 by jantje