Sloeber/arduino-eclipse-plugin #827: ATTinyCore - Sloeber won't change some settings (clock/B.O.D/Timer1 Clock)

liebman, 2017-09-20:
I found that changing the "Clock", "B.O.D" and "Timer 1 Clock" settings has no effect:
Here is the form I submitted (note I've set clock to 1 MHZ internal):
[image]

And here is the result opening properties again:
[image]

jantje, 2017-09-21:
I just tested here on the nightly and it works fine.
There has been some changes in this area because for upload to be working fine.
Did you just switched to the nightly? Because then Sloeber should have lost all settings but once you have re entered them all should be fine.
Just tested here with your setup.

liebman, 2017-09-21:
Ok, I updated to a more recent nightly and its working fine now!  Thanks again.

closed 2017-09-21 by liebman