Sloeber/arduino-eclipse-plugin #828: ATTinyCore avrdude upload fails when sloeber is running

liebman, 2017-09-23:
Mac OSX & nightly: `Sloeber	4.1.0.201709180107	io.sloeber.feature.feature.group	Jan Baeyens and Others`
If I have eclipse (sloeber) running avrdude fails like this:
```
flugelberry:AnalogClock liebman$ /Users/liebman/Applications/cpp-neon/Eclipse.app/Contents/Eclipse//arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -C/Users/liebman/Arduino/hardware/ATTinyCore/avr/avrdude.conf -pattiny85 -cusbtiny -Uflash:w:/Users/liebman/src/AnalogClock/I2CAnalogClock/Release/I2CAnalogClock.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e930b (probably t85)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/Users/liebman/src/AnalogClock/I2CAnalogClock/Release/I2CAnalogClock.hex"
avrdude: writing flash (5614 bytes):

Writing | ########                                           | 15% 1.36s
avrdude: error: usbtiny_send: Broken pipe (expected 64, got -32)
Writing | ################################################## | 100% 11.12s

avrdude: 5614 bytes of flash written
avrdude: verifying flash memory against /Users/liebman/src/AnalogClock/I2CAnalogClock/Release/I2CAnalogClock.hex:
avrdude: load data flash data from input file /Users/liebman/src/AnalogClock/I2CAnalogClock/Release/I2CAnalogClock.hex:
avrdude: input file /Users/liebman/src/AnalogClock/I2CAnalogClock/Release/I2CAnalogClock.hex contains 5614 bytes
avrdude: reading on-chip flash data:

Reading | #                                                  | 2% 0.16savrdude: 2 retries during read
Reading | #########                                          | 17% 1.42savrdude: 2 retries during read
Reading | #############                                      | 26% 2.23savrdude: 1 retries during read
Reading | ##########################                         | 51% 4.09savrdude: 1 retries during read
Reading | #######################################            | 77% 6.03savrdude: 1 retries during read
Reading | ########################################           | 80% 6.36savrdude: 1 retries during read
Reading | ################################################## | 100% 7.82s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0x08 != 0x39
avrdude: verification error; content mismatch

avrdude: safemode: Fuses OK (E:FF, H:DF, L:62)

avrdude done.  Thank you.
```

If I exit slobber then it succeeds:
```
flugelberry:AnalogClock liebman$ /Users/liebman/Applications/cpp-neon/Eclipse.app/Contents/Eclipse//arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -C/Users/liebman/Arduino/hardware/ATTinyCore/avr/avrdude.conf -pattiny85 -cusbtiny -Uflash:w:/Users/liebman/src/AnalogClock/I2CAnalogClock/Release/I2CAnalogClock.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e930b (probably t85)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file "/Users/liebman/src/AnalogClock/I2CAnalogClock/Release/I2CAnalogClock.hex"
avrdude: writing flash (5614 bytes):

Writing | ################################################## | 100% 8.54s

avrdude: 5614 bytes of flash written
avrdude: verifying flash memory against /Users/liebman/src/AnalogClock/I2CAnalogClock/Release/I2CAnalogClock.hex:
avrdude: load data flash data from input file /Users/liebman/src/AnalogClock/I2CAnalogClock/Release/I2CAnalogClock.hex:
avrdude: input file /Users/liebman/src/AnalogClock/I2CAnalogClock/Release/I2CAnalogClock.hex contains 5614 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 8.75s

avrdude: verifying ...
avrdude: 5614 bytes of flash verified

avrdude: safemode: Fuses OK (E:FF, H:DF, L:62)

avrdude done.  Thank you.
```

liebman, 2017-09-23:
Ok - thats not 100% accurate.  But I find that avrdude fails sometimes when run from the IDE and after it fails it will fail when run from a terminal window until you quit the IDE.

jantje, 2017-09-23:
Sounds like the serial port is not closed properly
Do you have the serial monitor open when the first fail happens?

liebman, 2017-09-23:
This is using the programmer (usbtiny) not the serial.

rlogiacco, 2017-09-23:
But do you have the serial window open when attempting to program with the programmer?

liebman, 2017-09-23:
Nope.  But now its not happening.  I'm wondering if there was a failed upload and somehow the avrdude process did not completely exit.  If this does not re-occure in the next few days I'll close the ticket.

liebman, 2017-09-23:
FYI - I'm programming an ATtiny85 - no serial as I'm using all of the pins including setting the fuse to make the reset pin an IO pin.

liebman, 2017-09-24:
Ok - I retract what I said about not using serial.   I'm not using serial on the ATtiny85 (its a usb programmer) however I do use serial to an ESP8266....  It seems if the IDE has that serial port open then avrdude has problems with finding or talking with the programmer for the ATtiny85.  I'm specifically using this one: https://www.sparkfun.com/products/9825.

liebman, 2017-09-24:
I'll close this.  It seems to be an issue with avrdude.  If I connect to the ESP8266 serial with "screen" and run avrdude it fails.

closed 2017-09-24 by liebman

jantje, 2017-09-24:
thanks for looking into this and feeding back :-)
Seems like Sloeber is currently not only supports but is also used with a lot of non-arduino boards :-)