Sloeber/arduino-eclipse-plugin #829: Cannot create ESP8266 project

hundsmiachn, 2017-09-25:
Hi

First of all thanks for your time and efforts to create this package. It really saves a lot of time and makes development much more fun than with the original arduino IDE....

I installed the latest package (V4.1) and added the esp8266 package board link.
When I try to create a project, it shows me the ESP8266 boards and settings.
Regardless what I choose, the next and finish buttons are always grayed out, I cannot save the new project. 

It is the same, when I create a project with an arduino board and change the a board afterwards. Here I can create the project, but when I change the board to an ESP8266 board the OK and APPLY button are grayed out, thus I cannot store the board settings. I think it is a problem with the Frequency setting of the esp boards.

I am already using V3.0 for some time now.

thanks
regards
Erich

hundsmiachn, 2017-09-25:
Tried a different esp8266 link, works now. Sorry for the issue

http://eclipse.baeyens.it/download/package_esp8266com_index.json

closed 2017-09-25 by hundsmiachn

jantje, 2017-09-25:
thanks for the info. which esp json did you use first?

hundsmiachn, 2017-09-25:
Hi

I used the standard link
http://arduino.esp8266.com/stable/package_esp8266com_index.json

regards
Erich

jantje, 2017-09-25:
I just see you are using V3
That is old.
In V4 http://arduino.esp8266.com/stable/package_esp8266com_index.json automatically refers to http://eclipse.baeyens.it/download/package_esp8266com_index.json as you can see in #745 

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje