Sloeber/arduino-eclipse-plugin #830: Doing Serial.println every frame lags the Sloeber

Silidrone, 2017-09-25:
Whenever you send something to be printed in Serial Monitor, the whole sloeber lags and cant even compile...

jantje, 2017-09-25:
Can you be more specific on what you do and why you think this is worth a issue?
Also which os/hardware do you use, which sloeber version which board?

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jantje, 2017-10-08:
closing due to lack of response.

Silidrone, 2017-10-10:
I am sorry for not responding this long, I am using esp8266, and when I open serial monitor, if I print too much in my code, the sloeber delays every action such as undo, build, upload... I am on Windows 10, CPU: i7-4710HQ @ 2.50 GHz, RAM: 16GB.  It is 4.1 Version, I think, I am not sure how to check which version it is.

jantje, 2017-10-10:
>if I print too much in my code,

There is indeed a processor penalty for using serial as there is a penalty for using any device.
You may not realize but lowering the baud rate will make your esp suffer the penalty as the serial queue will get full.
So the question is how much do you send and is the processor penalty caused by Sloeber for receiving that amount of data acceptable or not.

Silidrone, 2017-10-10:
The esp8266 lags when you send him more then 10000 prints at once, which is not sloeber's problem, but esp8266's. But forget that, lets say I dont send that much at once, lets say I send 200 at once every frame (in loop), sloeber has (serial monitor in sloeber) hard time recieveing that much, and delays my all actions, as I said. By the way, the baud rate is set at 9600.

uzi18, 2017-10-10:
Have also encounter slow printing in serial monitor and terminal.

Even some upload tools does not work properly in terminal but same command
pasted to real terminal works ok.

Silidrone, 2017-10-10:
By the way, is the issue reopen-able? As I did respond now.

jantje, 2017-10-10:
From what I understand there are multiple requests to increase the performance of the serial monitor.
However some realism is needed.
1) Sloeber will never be as performant as a real terminal program.
2) lag is normal in a serial communication where there is a output and a input queue. If you want to know the "pc/sloeber" lag you need to make sure the "output buffer" on the embedded device is empty before sending the message used to time the lag. This is code like I have in my library here https://github.com/jantje/ArduinoLibraries/blob/master/SerialCommunicator/SerialCommunicator.cpp#L168
3) When you need performance do not filter for the plotter. This filtering has serious impact on the performance.

@uzi18 you say
>Even some upload tools does not work properly in terminal but same command

What do you mean by that? I do not see the link between upload tools and the terminal. When Sloeber starts upload tools it closes  the terminal connection runs the upload tool and reopens the connection.

uzi18, 2017-10-10:
Ad.1 if it is ok for you so lagging is not a problem.
My laptop is not new one but in Arduino IDE it works quiet fast. In the
same conditions.

For programming with script megaflash/esp-link, need own programers.txt
file, it is almost work -  once per 5-6 launches.
I need investigate it but in different issue.
Not all options are passed to executable, this is related to
platform.txt/programmers.txt.

   https://github.com/jantje/ArduinoLibraries/blob/master/
   SerialCommunicator/SerialCommunicator.cpp#L168

jantje, 2017-10-10:
>For programming with script megaflash/esp-link, need own programers.txt file, it is almost work - once per 5-6 launches. I need investigate it but in different issue. Outside Sloeber it works every time. Another issue not all options are passed to uploader executable, this is related to platform.txt/programmers.txt support.

this is a different issue and is not related to serial monitor performance.

jantje, 2017-10-20:
I have been thinking about this one. I fixed some delay issues in closing the serial monitor which may be related to upload issues because of the delays the serialmonitor causes due to lag after closing.
So the nightly might behave better.

jantje, 2017-11-28:
The nightly from 29/11 should be far better.
Can you do a test to see whether it is sufficient?

uzi18, 2017-11-28:
I will try it,  thanks for info

jantje, 2017-12-11:
did you have time to try?

jantje, 2018-01-07:
Please reopen if not fixed

closed 2018-01-07 by jantje