Sloeber/arduino-eclipse-plugin #833: boards.txt not foujd

laica10, 2017-09-26:
Just installed sloeber complete pack on macOS I suppose it's version 4 but when I start sloeber it just gives me an error about wich says: "No boards.txt files found in the arduino hardware folders". 
Someone encountered this problem or solved it yet?

jantje, 2017-09-27:
Can you provide a image of the error.
Have you looked at the video's?
https://www.youtube.com/watch?v=zc2JLbtbI7Q&list=PLCSjf7QmP3TDPSjT_5UIp0DHXzKdhIQsF

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

seamaster101, 2017-09-29:
I have similar problem with teensy boards. They are not installed by default, and even when I follow old instructions, it shows up, in the board selection, the is not upload protocol selection available and will not let me create teensy project
[image]

jantje, 2017-09-29:
@seamaster101 I think you need to enlarge the window as there are more options which have not yet been set.
As to the uploader you will have to copy the programmers.txt as explained in #826.
As to the teensy install: what do you mean with "and even when I follow old instructions'

uzi18, 2017-09-29:
@jantie: is it possible to add there always shown slider as it sometimes
hide some options below in this window.

jantje, 2017-09-29:
If you know how, feel free.

uzi18, 2017-09-29:
Ok will investigate it.

jantje, 2017-09-29:
It should be somewhere in here https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.ui/src/io/sloeber/ui/project/properties/BoardSelectionPage.java

laica10, 2017-09-29:
[image]
This is the error I'm having, sorry for the delayed response but I was really busy.

mjmeijer, 2017-09-29:
Reinstall and move Sloeber to Applications folder first before launching fir first time. Before launching enable Install applications from anywhere in Security preference

laica10, 2017-09-29:
@mjmeijer, Thanks! It worked with just moving Sloeber to applications. Note that u can't enable install applications from anywhere anymore in security preferences.

closed 2017-09-29 by mjmeijer

seamaster101, 2017-09-30:
@jantje  Reporting back that my issue was resolved too. There is more hidden window, but not an indication "like a scroll-bar" that some part can not be seen.  
I also copied programmers.txt as indicated above and that resolved "upload protocol: default" only issue.
I only need to find out how to make my existing Teensy project compile. Right now there is no Arduino option in the properties of imported project, but issue for another day and it is not related to the above,
thank you so much!