Sloeber/arduino-eclipse-plugin #835: All library manager functions stopped working

winginitau, 2017-10-02:
Great project - will be pleased to contribute bug testing and financially. 

For some reason all library functions have stopped working. Menu item "Add library.." does nothing. "Reattach..." does nothing. Build all reports:
 
(An internal error occurred during: "Adding Arduino libs...".
0)

In my project I have an unresolved library include but cannot get these functions to behave. 

This is with a fresh install of v4, after having created a new .ino project, pasted in the main code and then moved in a bunch of local includes. 

Many thanks
Brendan

jantje, 2017-10-02:
My first idea is that this could be a duplicate of #755 

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jantje, 2017-10-08:
Is there a update?

winginitau, 2017-10-09:
Yes. I think a repeat or something similar. I manually removed all folders under <project>/Release/libraries and the problem cleared and the sub-folders were rebuilt. Thanks heaps for your help. I'm a newbie with eclipse but starting to get my head around it.....

closed 2017-10-09 by winginitau