Sloeber/arduino-eclipse-plugin #837: Unable to upload code on Win10 via USB

lukasadamek, 2017-10-06:
Hello
I have installed last Sloeber 4.1.0.201705252335 on PC with Windows 10, but I am not able to upload the program successfully. I am trying example WebServer.

The build is showing success.
```javascript
20:10:05 **** Incremental Build of configuration Release for project WebServer ****
"D:\\Docasne\\Sloeber\\arduinoPlugin\\tools\\make\\make" 
'Building file: ..\WebServer.ino'
'Starting C++ compile'
"D:\Docasne\Sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   -I"D:\Docasne\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\cores\arduino" -I"D:\Docasne\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\variants\mega" -I"D:\Docasne\Sloeber\arduinoPlugin\libraries\Ethernet\1.1.2" -I"D:\Docasne\Sloeber\arduinoPlugin\libraries\Ethernet\1.1.2\src" -I"D:\Docasne\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\libraries\SPI" -I"D:\Docasne\Sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\libraries\SPI\src" -MMD -MP -MF"WebServer.d" -MT"WebServer.o" -D__IN_ECLIPSE__=1 -x c++ "..\WebServer.ino"  -o  "WebServer.o"
'Finished building: ..\WebServer.ino'
' '

20:10:06 Build Finished (took 901ms)
```

The upload is also showing success, but actually the Arduino MEGA 2560 is not programmed.
```javascript
Starting upload
using arduino loader

LaunchingD:\Docasne\Sloeber\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -CD:\Docasne\Sloeber\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/etc/avrdude.conf -patmega2560 -cstk500v2 -Pusb -Uflash:w:D:\Lukas\Vyvoj\Workspace - Arduino2\WebServer/Release/WebServer.hex:i 
Output:

avrdude done.  Thank you.

/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude finished
upload done
```

[image]
[image]
[image]
[image]

jantje, 2017-10-06:
Have you tried the default upload protocol?

jantje, 2017-10-08:
is there a update?

lukasadamek, 2017-10-08:
Hello, I tried Default Upload Protocol and upload to arduino works. 
Thanks for help.

jantje, 2017-10-08:
ok closing the issue

closed 2017-10-08 by jantje