Sloeber/arduino-eclipse-plugin #838: mvn build verify fails with compilation failure

NetScr1be, 2017-10-08:
uname -a returns;

Linux AspireOneLinux 3.16.0-4-686-pae #1 SMP Debian 3.16.7-ckt25-1 (2016-03-06) i686 GNU/Linux

Maven version

Version: 3.3.9-3\~bpo8+1
Installed-Size: 113
Maintainer: Debian Java Maintainers <pkg-java-maintainers@lists.alioth.debian.org>
Architecture: all
Replaces: maven2 (<< 2.2.1-11)
Depends: default-jre-headless (>= 2:1.7) | java7-runtime-headless, libmaven3-core-java (= 3.3.9-3~bpo8+1)
Breaks: maven2 (<< 2.2.1-11)

> 

[ERROR] Failed to execute goal org.eclipse.tycho:tycho-compiler-plugin:0.24.0:compile (default-compile) on project io.sloeber.core: Compilation failure: Compilation failure:

[ERROR] /home/patrick/code/arduino-eclipse-plugin/io.sloeber.core/src/io/sloeber/core/common/InstancePreferences.java:[122]

[ERROR] setGlobalValue(KEY_PRIVATE_LIBRARY_PATHS, String.join(File.pathSeparator, folderName));
[ERROR] ^^^^
[ERROR] The method join(String, String[]) is undefined for the type String
[
ERROR] /home/patrick/code/arduino-eclipse-plugin/io.sloeber.core/src/io/sloeber/core/common/InstancePreferences.java:[134]
[ERROR] setGlobalValue(KEY_PRIVATE_HARDWARE_PATHS, String.join(File.pathSeparator, folderName));
[ERROR] ^^^^
[ERROR] The method join(String, String[]) is undefined for the type String

[ERROR] /home/patrick/code/arduino-eclipse-plugin/io.sloeber.core/src/io/sloeber/core/api/BoardsManager.java:[338]
[ERROR] Messages.Helpers_No_boards_txt_found + String.join("\n", hardwareFolders), null)); //$NON-NLS-1$
[ERROR] ^^^^
[ERROR] The method join(String, String[]) is undefined for the type String

[ERROR] /home/patrick/code/arduino-eclipse-plugin/io.sloeber.core/src/io/sloeber/core/api/BoardsManager.java:[435]
[ERROR] this.boards = String.join("\n", internalPlatformm.getBoardNames()); //$NON-NLS-1$
[ERROR] ^^^^
[ERROR] The method join(String, List<String>) is undefined for the type String

[ERROR] /home/patrick/code/arduino-eclipse-plugin/io.sloeber.core/src/io/sloeber/core/api/VersionNumber.java:[40]
[ERROR] return String.join(".", this.parts); //$NON-NLS-1$
[ERROR] ^^^^
[ERROR] The method join(String, String[]) is undefined for the type String

[ERROR] 5 problems (5 errors)
<

Please advise on any other info needed and how to proceed

jantje, 2017-10-08:
This is probably a duplicate of #334

wimjongman, 2017-10-08:
You could have figured this out yourself.

**The method join(String, String[]) is undefined for the type String** because you compile with Java 7 and we code with Java 8.

Install Java 8 and try again.

> [ERROR] return String.join(".", this.parts); //$NON-NLS-1$
> [ERROR] ^^^^
> [ERROR] The method join(String, String[]) is undefined for the type String

closed 2017-10-08 by wimjongman