Sloeber/arduino-eclipse-plugin #839: How to remove -fno-rtti flag, I want to use dynamic_cast

Silidrone, 2017-10-10:
I have virtual inheritance, so in order to cast Base class to Derived class, I need to use dynamic_cast, because the base is virtual. So, the point is, I need to use dynamic_cast. But I am not able to. It gives me an error: "error: 'dynamic_cast' not permitted with -fno-rtti". So, what should I do? I tried to change "A.COMPILER.CPP.FLAGS" environment variable as it has "-fno-rtti" flag, so I just removed the flag, and restarted the Sloeber, but it just ignores my changes, and reverts back to old state (adding "-fno-rtti"). I also tried selecting the option "Replace native environment with specified one" and then changing, still, it ignores my changes...

jantje, 2017-10-10:
As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

As the A. implies: this environment variable is Arduino controlled as such it will be overwritten by times.
You can however change the value of compiler.cpp.flags in the platform.txt file. As long as you don't upgrade or reinstall the change should be permanent.

closed 2017-10-10 by jantje

Silidrone, 2017-10-10:
Thank you really much! It solved the problem!

Silidrone, 2017-10-10:
Ooops, it did solve the problem, but now, another problem occured (which I think has to do with this, when I changed, this error occured), the error says (I totally have no idea what this means): Description	Resource	Path	Location	Type
D:\myproject/Release/myproject.elf section `.text' will not fit in region `iram1_0_seg'.