Sloeber/arduino-eclipse-plugin #840: I get this error: '.text' will not fit in region 'iram1_0_seg'. how to solve this?

Silidrone, 2017-10-10:
I've just got this error after compiling my code in Sloeber (OS: WINDOWS 10): `D:\myproject/Release/myproject.elf section '.text' will not fit in region 'iram1_0_seg'. 
The error in console: c:/sloeber/arduinoplugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: D:\myproject/Release/myproject.elf section '.text' will not fit in region 'iram1_0_seg'.`

jantje, 2017-10-10:
Are you a patron of mine? Because this is not related to sloeber at all.

closed 2017-10-10 by jantje

Silidrone, 2017-10-10:
I am not, does donating instead of being patron count as a contributor?

Silidrone, 2017-10-10:
By the way, this is related to Sloeber, as you said if I want to remove -fno-rtti flag I need to edit platform.txt (on my other issue), and yeah, it has been removed, but the moment I change the platform.txt and build, I get this error. So, I think it is related to Sloeber.

jantje, 2017-10-10:
-fno-rtti flag is a compile optimisation that makes your program smaller.
As far as I see what is going on is that once you removed this optimisation your program is to big to fit in some part of your embedded device.
And no it is not Sloeber related as Sloeber executes the command as provided.