Sloeber/arduino-eclipse-plugin #843: Error when passing ENUM as a function parameter.

helltp, 2017-10-11:
In my main ino file i have got an enumeration llike this:
`enum Color {Red, Green,Blue}`
And a function that has a Color type parameter:
`void setColor(Color newColor){...}`

When I try to build it throws an error:

> ..\\.ino.cpp:17:19: error: 'Color' was not declared in this scope

I checked the ino file, although it says that it is generated and it should not be modified, if I change the function prototype from void `setColor(Color newColor);` to void `setColor(int newColor);` it works.

jantje, 2017-10-11:
put your 
`enum Color {Red, Green,Blue}`
in a header file

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

helltp, 2017-10-13:
Thank you for the help and the answer. 

As a senior softwer engineer student I'm considering to help you in my spare time.

closed 2017-10-13 by helltp