Sloeber/arduino-eclipse-plugin #846: SunCertPathBuilderException: unable to find valid certification path to requested target

barrio, 2017-10-13:
I'm on Ubuntu 16.04 using openjdk-9-jre.

I tried to download source with:
`mvn clean -Dtest=RegressionTest verify`
And get this error:
[INFO] Scanning for projects...
Downloading: https://repo.maven.apache.org/maven2/org/sonatype/oss/oss-parent/7/oss-parent-7.pom
[ERROR] [ERROR] Some problems were encountered while processing the POMs:
[FATAL] Non-resolvable parent POM for com.github.brodykenrick.arduino-eclipse-plugin:root:2.2.0: Could not transfer artifact org.sonatype.oss:oss-parent:pom:7 from/to central (https://repo.maven.apache.org/maven2): sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target and 'parent.relativePath' points at wrong local POM @ line 9, column 10
 @ 
[ERROR] The build could not read 1 project -> [Help 1]
[ERROR]   
[ERROR]   The project com.github.brodykenrick.arduino-eclipse-plugin:root:2.2.0 (/home/barrios/arduino-eclipse-plugin/pom.xml) has 1 error
[ERROR]     Non-resolvable parent POM for com.github.brodykenrick.arduino-eclipse-plugin:root:2.2.0: Could not transfer artifact org.sonatype.oss:oss-parent:pom:7 from/to central (https://repo.maven.apache.org/maven2): sun.security.validator.ValidatorException: PKIX path building failed: sun.security.provider.certpath.SunCertPathBuilderException: unable to find valid certification path to requested target and 'parent.relativePath' points at wrong local POM @ line 9, column 10 -> [Help 2]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/ProjectBuildingException
[ERROR] [Help 2] http://cwiki.apache.org/confluence/display/MAVEN/UnresolvableModelException

Before I tried to build from source similar errors appeared during update process of all nightly builds (4.0.0 / 4.0.1 / 4.1.0 ) for linux 64 bit and also when trying to install [http://eclipse.baeyens.it/update/V4/nightly](sloeber) into an existing neon 3 installation.

mjmeijer, 2017-10-13:
I think the sonatype pom parent is obsolete and can be removed

barrio, 2017-10-14:
Sorry, but I'm a total Java / Maven noob. I commented out the parent tag in pom.xml ...

```
<?xml version="1.0" encoding="UTF-8"?>
<project xmlns="http://maven.apache.org/POM/4.0.0" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="http://maven.apache.org/POM/4.0.0 http://maven.apache.org/xsd/maven-4.0.0.xsd">
  <modelVersion>4.0.0</modelVersion>
  <groupId>com.github.brodykenrick.arduino-eclipse-plugin</groupId>
  <artifactId>root</artifactId>
  <version>2.2.0</version>
  <packaging>pom</packaging>
  <!--  
	<parent>
        <groupId>org.sonatype.oss</groupId>
        <artifactId>oss-parent</artifactId>
        <version>7</version>
        <relativePath>org.sonatype.oss:oss-parent</relativePath>
    </parent>
  -->
	<scm>
        <connection>scm:git:git://github.com/${github.repo.username}/arduino-eclipse-plugin.git</connection>
        <developerConnection>scm:git:git@github.com:${github.repo.username}/arduino-eclipse-plugin.git</developerConnection>
        <url>http://github.com/${github.repo.username}/arduino-eclipse-plugin</url>
    </scm>

    <!-- Sonatype repos -->
    <distributionManagement>
        <snapshotRepository>
            <id>sonatype-nexus-snapshots</id>
            <name>Sonatype Nexus snapshot repository</name>
            <url>https://oss.sonatype.org/content/repositories/snapshots</url>
        </snapshotRepository>
        <repository>
            <id>sonatype-nexus-staging</id>
            <name>Sonatype Nexus release repository</name>
            <url>https://oss.sonatype.org/service/local/staging/deploy/maven2/</url>
        </repository>
    </distributionManagement>
  
  <modules>
    <module>io.sloeber.parent</module>
  </modules>
</project>
```
... and still get errors:
~/arduino-eclipse-plugin (master) [i] ⋊> mvn clean -Dtest=RegressionTest verify
[INFO] Scanning for projects...
Downloading: https://repo.maven.apache.org/maven2/org/sonatype/oss/oss-parent/9/oss-parent-9.pom
[ERROR] [ERROR] Some problems were encountered while processing the POMs:
[FATAL] Non-resolvable parent POM for com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.parent:4.1.0-SNAPSHOT: Could not transfer artifact org.sonatype.oss:oss-parent:pom:9 from/to platform_oxygen (http://download.eclipse.org/eclipse/updates/4.7): Cannot access http://download.eclipse.org/eclipse/updates/4.7 with type p2 using the available connector factories: BasicRepositoryConnectorFactory and 'parent.relativePath' points at wrong local POM @ line 14, column 11
 @ 
[ERROR] The build could not read 1 project -> [Help 1]
[ERROR]   
[ERROR]   The project com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.parent:4.1.0-SNAPSHOT (/home/barrios/arduino-eclipse-plugin/io.sloeber.parent/pom.xml) has 1 error
[ERROR]     Non-resolvable parent POM for com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.parent:4.1.0-SNAPSHOT: Could not transfer artifact org.sonatype.oss:oss-parent:pom:9 from/to platform_oxygen (http://download.eclipse.org/eclipse/updates/4.7): Cannot access http://download.eclipse.org/eclipse/updates/4.7 with type p2 using the available connector factories: BasicRepositoryConnectorFactory and 'parent.relativePath' points at wrong local POM @ line 14, column 11: Cannot access http://download.eclipse.org/eclipse/updates/4.7 with type p2 using the available layout factories: Maven2RepositoryLayoutFactory: Unsupported repository layout p2 -> [Help 2]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/ProjectBuildingException
[ERROR] [Help 2] http://cwiki.apache.org/confluence/display/MAVEN/UnresolvableModelException

jantje, 2017-10-14:
>Sorry, but I'm a total Java / Maven noob.

Don't worry, I am to.
The only thing I can tell you is that travis builds on Ubuntu 14.04.5 LTS work fine.
https://travis-ci.org/Sloeber/arduino-eclipse-plugin/builds/287650319

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jantje, 2017-10-26:
@barrio 
Did you make any progress? Can you feedback?

barrio, 2017-10-28:
Only thing I can say is that the problem is the access to the following URLs:
https://raw.githubusercontent.com/jantje/ArduinoLibraries/master/jantje_library_index.json
https://raw.githubusercontent.com/jantje/hardware/master/package_jantje_index.json

When I press "Apply" in "Third party index url's" under "Arduino", the same error as described above is shown. But I can't say what exactly is the problem with the certification path to access those sites.

jantje, 2017-10-28:
I'm confused here.
The first comment was on maven build and the last comment is when being in the preferences

barrio, 2017-10-28:
Me too :-). As I wrote above, I have the problem when I'm using precompiled sloeber nightly builds after starting during update process and as well when I compile it on my own machine and as well when I use nightly plugin V4 with fresh eclipse neon CDT. So I think there is a problem of accessing those json files in your github repo in ALL of these different ways. In Firefox I can access them without any certificate warnings, so I don't know wether the Problem is related to my Java version (openjdk version "1.8.0_131") or something else related to my OS Ubuntu 16.04, which you said is different from yours, or whatever?

jantje, 2017-10-28:
I think there is somewhere a security setting issue on your system.

jantje, 2017-11-10:
Can this be closed?

barrio, 2017-11-19:
I got similar errors in IntelliJ when trying to update kotlin version in build.gradle, but in Android Studio 3 the same update runs fine. Anyway, that's my personal problem, Issue can be closed.

closed 2017-11-19 by barrio