Sloeber/arduino-eclipse-plugin #847: Arduino_Core_STM32 linker includes arduino.a twice

kaofishy, 2017-10-13:
Hi, 

When using the board files from the official STM32 Arduino core (https://github.com/stm32duino/Arduino_Core_STM32), compilation fails with many "multiple definitions" errors. Upon inspection, I discovered that the library archive arduino.a is included twice during the "combiner" phase. This is the link command Sloeber generates:

`"C:\sloeber\/arduinoPlugin/packages/STM32/tools/arm-none-eabi-gcc/6-2017-q2-update/bin/arm-none-eabi-gcc" -mthumb -Os -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DF_CPU=80000000L "-TC:\sloeber\/arduinoPlugin/packages/STM32/hardware/stm32/variants/NUCLEO_L432KC/ldscript.ld" "-Wl,-Map,C:\Users\tonyk\Documents\sloeber-workspace\test2/Release/test2.map"  -o "C:\Users\tonyk\Documents\sloeber-workspace\test2/Release/test2.elf" "-LC:\Users\tonyk\Documents\sloeber-workspace\test2/Release" -Wl,--start-group    .\.ino.cpp.o   C:\Users\tonyk\Documents\sloeber-workspace\test2\Release\arduino.ar   "-LC:\sloeber\/arduinoPlugin/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Lib/GCC/" -larm_cortexM4l_math -Wl,--whole-archive  C:/Users/tonyk/Documents/sloeber-workspace/test2/Release/arduino.ar  -Wl,--no-whole-archive -lc -Wl,--end-group -lm -lgcc -lstdc++ --specs=nano.specs`

while this is the link command the Arduino IDE generates:

`"C:\Users\tonyk\AppData\Local\Arduino15\packages\STM32\tools\arm-none-eabi-gcc\6-2017-q2-update/bin/arm-none-eabi-gcc" -mthumb -Os  -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -mcpu=cortex-m4 -DF_CPU=80000000L "-TC:\Users\tonyk\AppData\Local\Arduino15\packages\STM32\hardware\stm32\2017.9.22\variants\NUCLEO_L432KC/ldscript.ld" "-Wl,-Map,C:\Users\tonyk\AppData\Local\Temp\arduino_build_670349/sketch_oct13a.ino.map"   -o "C:\Users\tonyk\AppData\Local\Temp\arduino_build_670349/sketch_oct13a.ino.elf" "-LC:\Users\tonyk\AppData\Local\Temp\arduino_build_670349" -Wl,--start-group "C:\Users\tonyk\AppData\Local\Temp\arduino_build_670349\sketch\sketch_oct13a.ino.cpp.o" "C:\Users\tonyk\AppData\Local\Temp\arduino_build_670349\core\PeripheralPins.c.o" "C:\Users\tonyk\AppData\Local\Temp\arduino_build_670349\core\variant.cpp.o"  "-LC:\Users\tonyk\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Lib/GCC/" -larm_cortexM4l_math -Wl,--whole-archive "C:\Users\tonyk\AppData\Local\Temp\arduino_build_670349/..\arduino_cache_355286\core\core_STM32_stm32_Nucleo_32_pnum_NUCLEO_L432KC,upload_method_STLink,xserial_SerialAll,usb_none,opt_osstd_451bdd72148238307472646654d3f8b9.a" -Wl,--no-whole-archive -lc -Wl,--end-group -lm -lgcc --specs=nano.specs`

Note that there are two instances of arduino.a included in the command Sloeber generates.

This happens both with the latest release and the git version of the STM32 core.

jantje, 2017-10-14:
IMHO this is fixed in the nightly

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

kaofishy, 2017-10-14:
Okay thanks, I will test nightly and report back.

kaofishy, 2017-10-14:
Tried nightly and the bug has been fixed.

closed 2017-10-14 by kaofishy