Sloeber/arduino-eclipse-plugin #848: MDNS packet flood using plugin when multiple OTA clients exist

kurtgo, 2017-10-16:
I recently deployed 7 devices with OTA Update enabled (they respond to mdns _arduino._tcp.local queries.
Each copy of eclipse blasts 6 or 7 MDNS broadcast messages (like this wireshark capture)
2963	758.659438000	192.168.1.199	224.0.0.251	MDNS	79	Standard query 0x0000  PTR _arduino._tcp.local, "QM" question

Although they seem benign, each query gets 7 responses (one from each arduino running OTA).  This flood is replicated numerous times, and eventually causes the entire network to clog with mDNS query/result packets.

Arduino IDE also causes the same behavior, so I would bet it is related to the jmdns

Steps to reproduce:
1) fire up many OTA update arduino devices on the same subnet connected to the same wireless router
2) start multiple copies of eclipse and/or Arduino IDE.
3) start wireshark and monitor mDNS queries to see the flood.

The cause is probably related to the use of the jmdns package, however the mDNS standard requires broadcasters listen for mDNS requests, and not re-broadcast.  Since each copy of eclipse is emitting a broadcast for each OTA device.

jantje, 2017-10-16:
Sounds like you know what you are doing.
But I have no clue what you are saying.
As Arduino IDE has far more resources and you claim they have the same problem I would try to bang their door.
But then again, I have no clue what you are talking about.
Jantje

kurtgo, 2017-10-16:
Well, I entered a ticket there too.  Apparently, this issue has to do with bugs and deficiencies in the jmdns package.  Updating jmdns to 3.5.4 solves some of the issue, as it does not send nearly as many mDNS queries when it sends the resolve multi-cast messages (now it's only 3, so that's lots better).

Apparently the Arduino core code that you are using in Slobber is just copied from Arduino core.

I would like to update the jmdns package on my version of eclipse, but I have no clue about how to start that process on this git repo.  Should I fork the repo and update the jmnds package?

wimjongman, 2017-10-17:
See #655

closed 2017-10-17 by wimjongman

jantje, 2017-10-20:
@kurtgo 
The jmdns package has been upgraded to the latest version.
I hope this eases the pain.

wimjongman, 2017-10-23:
@kurtgo can you run your use case again?

kurtgo, 2017-10-28:
Updating jmdns will help, but that is only half the problem (albeit the more important half).  This issue https://github.com/arduino/Arduino/issues/6832 tracks the whole problem. It was fixed in Arduino core libraries so you should really update to the newer core code.