Sloeber/arduino-eclipse-plugin #850: Failed to create Projec on ESP8266 2.3.0  with Sloeber 4.1.0.201710131716

ghislainborremans, 2017-10-18:
When i use the 2.2.0 version i can create a sample project. with 2.3.0 it fails ??? 
any ideas ?

jantje, 2017-10-18:
@ghislainborremans 
Thanks for being a sponsor.
I just tried here using http://arduino.esp8266.com/stable/package_esp8266com_index.json version 2.3.0 and the default.ino
project creation and buils worked without any problems
[image]

did you get a error of some kind? is there a error in the error log?

ghislainborremans, 2017-10-18:
In the log it only says : failed to create project . 
I try to run the FSBrowserNG https://github.com/gmag11/FSBrowserNG  example UserConfigExample but i keep running into errors and that is why i try the 2.3.0.. 
I probable will have to start with a virgin situation because i think libraries are messed up somehow. 
I keep you posted.

jantje, 2017-10-18:
There is no stacktrace in the log with the error?
The libraries should not cause this. Have you tried uninstalling and reinstalling 2.3.0?

ghislainborremans, 2017-10-19:
I reinstalled 2.3. 0 and it works again

closed 2017-10-19 by ghislainborremans