Sloeber/arduino-eclipse-plugin #851: ESP8266:  unknown type name 'time_t'    time_t      tv_sec;

ghislainborremans, 2017-10-19:
When compiling the FSBrowserNG UserConfigExample.ino i get the following error. 
it  looks as it refers to the time.c file un under the core library under Sloeber.

D:\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\time.c:26:3: error: unknown type name 'time_t'
   time_t      tv_sec;

Thanks in advance for the help.

jantje, 2017-10-19:
check your cdt indexer options #825 
Check the libraries Sloeber has added.
Know that the nightly auto  installs libraries which (combined with the cdt explorative options) may lead to unwanted included libraries in your project.

ghislainborremans, 2017-10-19:
I have updated the indexer options  as indicated but n success.
I do not see extra libraries neither ?

jantje, 2017-10-19:
After having changed the indexer options you need to delete all the libraries in the libraries folder of the project so sloeber will import them again (hopefully this time the good ones).
If the problem persists after this, check whether Sloeber imported the good libraries

ghislainborremans, 2017-10-20:
Hello,
the procedure above works. 

The error is due to the fact that for the FSBrowserNG library it is necessary to copy the "time.h" file to "_time.h"  in order to call the "#include <_time.h> " statement from WebHandlerImpl.h ( see explanation in Troubleshooting section at https://github.com/gmag11/FSBrowserNG. This is needed because the project also uses Time.h (Timelib.h ).

It is important that you change the "#include time.h" to "#include _time.h" also in "time.c" file as well. this solved the issue. 

The files "time.c" and "time.h" are located in the  Sloeber/Arduino plugin folder ( ".\sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include"  and not in the Arduino1.5 folder as mentioned in Troubleshooting section at https://github.com/gmag11/FSBrowserNG..

closed 2017-10-20 by ghislainborremans

jantje, 2017-10-20:
:-) I thought it would be something like this.