Sloeber/arduino-eclipse-plugin #853: Error "make: *** [arduino/CDC.cpp.o] error1"

Nonth8031, 2017-10-26:
Hello:

I downloaded Eclipse Luna (using Java 1.7) and V2 plugin but it doesn't work. I install AVRDude plugin (and the AVR software) and Arduino IDE 1.6.0 too, I just do it because I had installed 1.8.3 (think that is the newest) but It was unsopported. My problems were 2, but the first I think that I fixed:
 1. The first one was that Eclipse didn't find "avr-gcc" and "avr-g++" (something like that, I'm not sure about the name). I solved it changing the directory where to find it, and they don't appear more.
2. The second is that I try to compile the program, a very simple program, a typical Blinking LED 13, but in Problems tab appears me "make: *** [arduino/CDC.cpp.o] error1". At the beginning, It was error 327 but, as you can see, I solved it (I hope it, at least).

Should I download Java 8 and V4 plugin? What is the problem? I don't know how to solve it, I spent many hours searching a solve but I didn't find anything.

I apologize for my English but I hope that you can understand the explain.
Thank you for your time 😄

jantje, 2017-10-26:
V2 is a very old version and we only support the latest version.
Please install the V4 product for your OS.

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

closed 2017-10-26 by jantje

ichigo92, 2017-11-22:
Hi @jantje,
I am using v4.2 sloeber, for some reason getting this same error.
`make: *** [sloeber.ino.cpp.o] Error 1`

[image]

[image]

Any help would be appreciated.

jantje, 2017-11-22:
V2 is not supported. So upgrade to the latest version