Sloeber/arduino-eclipse-plugin #855: *** No rule to make target

Nonth8031, 2017-10-30:
Hi again

I installed Sloeber V4 product but when I try to build (even an example) give this error:

10:17:36 **** Incremental Build of configuration Release for project ej ****
"C:\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
make: *** No rule to make target 'C:\Users\JM\My', needed by 'ej.elf'.  Stop.

10:17:38 Build Finished (took 2s.346ms)
 
How can I fix this error? I don't know what can be wrong.

Thank for your time
PS: I installed Java 8 before the Sloeber V4 product

jantje, 2017-10-30:
Did you install V4 or V4.1?
Can you provide some screenshots or video?
This way I can make a response video like this one https://www.youtube.com/watch?v=pcQv10LyIMY

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

Nonth8031, 2017-10-30:
I installed V4.1.
I have recorded and uploaded a video:  [video]( https://youtu.be/jrljZ7g5dHk)
Thank you

jantje, 2017-10-30:
You have spaces in your path :-)

Nonth8031, 2017-10-30:
How can I fix it? I'm not be able to find it :(

closed 2017-10-31 by wimjongman