Sloeber/arduino-eclipse-plugin #856: avrdude can't open correct COM device, but works in Arduino IDE

rtwfroody, 2017-10-31:
I'm using an Adafruit Feather 32u4 Radio with RFM69HCW Module.

Using the Arduino IDE, I can download and run a program on COM1. Using sloeber, when I go to download I get the following output:
```
Starting upload
using arduino loader
Forcing reset using 1200bps open/close on portCOM1
PORTS { COM1,} / { COM1,} => {}
PORTS { COM1,} / { COM1, COM5,} => { COM5,}
Comport reset took500ms
Using comportCOM5 from now onwards
Ending reset

LaunchingC:\sloeber\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -CC:\sloeber\/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/etc/avrdude.conf -patmega32u4 -cavr109 -PCOM1 -b57600 -D -Uflash:w:C:\Users\tnewsome\Synced\projects\thermostat\eclipse-workspace\sensor_v1/Release/sensor_v1.hex:i 
Output:
avrdude: ser_open(): can't open device "\\.\COM1": The system cannot find the file specified.

avrdude done.  Thank you.

/arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude finished
upload done
```

First, I'm confused about the message regarding COM5. Second, it seems to fail on COM1 even though that is the correct port.

I've been using this setup with no problem for some time with an Adafruit HUZZAH32 - ESP32 Feather, on COM3.

jantje, 2017-10-31:
Duplicate of #772 

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

closed 2017-10-31 by jantje

rtwfroody, 2017-10-31:
#772 seems different, but this might be a duplicate of #798, or potentially #778.

I'll try the nightly later.

rtwfroody, 2017-10-31:
Confirmed that this problem is fixed in the nightly build. I did have to reselect the Arduino board (to the same settings it had) to make it work.

jantje, 2017-10-31:
That is part of the standard upgrade procedure http://eclipse.baeyens.it/how_to.shtml#/n