Sloeber/arduino-eclipse-plugin #857: nightly: cannot open linker script file -Wl,--gc-sections

rtwfroody, 2017-10-31:
I updated to the nightly to get the fix for #856. Now my older project will no longer build:
```
'Starting combiner'
"C:\sloeber\/arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc" -g -Wall -Wextra -Os -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static "-LC:\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lib" "-LC:\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/ld" "-T" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy -o "C:\Users\tnewsome\Synced\projects\thermostat\eclipse-workspace\head-v1/V1/head-v1.elf" -Wl,--start-group    .\clockpage.cpp.o .\directpage.cpp.o .\gui.cpp.o .\mainpage.cpp.o .\state.cpp.o .\thermostat.cpp.o .\utils.cpp.o  .\libraries\Wire\Wire.cpp.o  .\libraries\WiFiManager\WiFiManager.cpp.o  .\libraries\SPI\SPI.cpp.o  .\libraries\SD\src\utility\Sd2Card.cpp.o .\libraries\SD\src\utility\SdFile.cpp.o .\libraries\SD\src\utility\SdVolume.cpp.o  .\libraries\SD\src\File.cpp.o .\libraries\SD\src\SD.cpp.o  .\libraries\RTClib\RTClib.cpp.o  .\libraries\JPEGDecoder\src\JPEGDecoder.cpp.o .\libraries\JPEGDecoder\src\picojpeg.c.o  .\libraries\ESP8266mDNS\ESP8266mDNS.cpp.o  .\libraries\ESP8266WiFi\src\ESP8266WiFi.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiMulti.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp.o .\libraries\ESP8266WiFi\src\ESP8266WiFiScan.cpp.o .\libraries\ESP8266WiFi\src\WiFiClient.cpp.o .\libraries\ESP8266WiFi\src\WiFiClientSecure.cpp.o .\libraries\ESP8266WiFi\src\WiFiServer.cpp.o .\libraries\ESP8266WiFi\src\WiFiUdp.cpp.o  .\libraries\ESP8266WebServer\src\ESP8266WebServer.cpp.o .\libraries\ESP8266WebServer\src\Parsing.cpp.o  .\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp.o  .\libraries\EEPROM\EEPROM.cpp.o  .\libraries\DNSServer\src\DNSServer.cpp.o  .\libraries\DHT_sensor_library\DHT.cpp.o .\libraries\DHT_sensor_library\DHT_U.cpp.o  .\libraries\ArduinoOTA\ArduinoOTA.cpp.o  .\libraries\Adafruit_STMPE610\Adafruit_STMPE610.cpp.o  .\libraries\Adafruit_MQTT_Library\Adafruit_MQTT.cpp.o .\libraries\Adafruit_MQTT_Library\Adafruit_MQTT_Client.cpp.o  .\libraries\Adafruit_ILI9341\Adafruit_ILI9341.cpp.o  .\libraries\Adafruit_GFX_Library\Adafruit_GFX.cpp.o .\libraries\Adafruit_GFX_Library\glcdfont.c.o  .\TFT_eSPI\Fonts\Font16.c.o .\TFT_eSPI\Fonts\Font32rle.c.o .\TFT_eSPI\Fonts\Font64rle.c.o .\TFT_eSPI\Fonts\Font72rle.c.o .\TFT_eSPI\Fonts\Font7srle.c.o .\TFT_eSPI\Fonts\glcdfont.c.o  .\TFT_eSPI\TFT_eSPI.cpp.o  .\HeatPump\src\HeatPump.cpp.o  .\GUIslice\src\GUIslice.c.o .\GUIslice\src\GUIslice_drv_adagfx.cpp.o .\GUIslice\src\GUIslice_drv_sdl.c.o .\GUIslice\src\GUIslice_drv_tft_espi.cpp.o .\GUIslice\src\GUIslice_ex.c.o   "C:\Users\tnewsome\Synced\projects\thermostat\eclipse-workspace\head-v1/V1/arduino.ar" -lm -lgcc -lhal -lphy -lpp -lnet80211 -lwpa -lcrypto -lmain -lwps -laxtls -lsmartconfig -lmesh -lwpa2 -llwip_gcc -lstdc++ -Wl,--end-group "-LC:\Users\tnewsome\Synced\projects\thermostat\eclipse-workspace\head-v1/V1" C:/Users/tnewsome/Synced/projects/thermostat/eclipse-workspace/head-v1/V1/arduino.ar
c:/sloeber/arduinoplugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: cannot open linker script file -Wl,--gc-sections: No such file or directory
collect2.exe: error: ld returned 1 exit status
makefile:126: recipe for target 'head-v1.elf' failed
make: *** [head-v1.elf] Error 1
```

Grepping around in .settings/org.eclipse.cdt.core.prefs finds mentions of `-T${A.BUILD.FLASH_LD}` which makes me think that that variable should be set but isn't. I tried adding it to the C/C++ Build Environment but that didn't help. I'm a bit stumped because I'm not familiar with how Eclipse manages projects and handles compilation.

jantje, 2017-10-31:
Please provide all information to reproduce.

rtwfroody, 2017-11-02:
I don't know what all you need. Attached is a workspace containing just the project that doesn't build, as well as the full compile log (or at least what the console showed when it had failed).
[workspace.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/1438819/workspace.zip)
[compile_log.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/1438820/compile_log.txt)

rtwfroody, 2017-11-02:
Note that this workspace is the way it is after attempting to build with the nightly. If I subsequently try to build with 4.1, it fails in the same way. So:

Starting with: 
[workspace-good.zip](https://github.com/Sloeber/arduino-eclipse-plugin/files/1438860/workspace-good.zip)

1. Build with 4.1: works
2. Build with nightly: fails. The workspace now looks like workspace.zip from the previous comment.
3. Build with 4.1: now also fails

rtwfroody, 2017-11-08:
Actually it's even worse than that. In step 2 above there's no need to build. Just opening the workspace with the nightly screws up the project to the point where it can no longer be built with either the nightly or with 4.1.

jantje, 2017-11-08:
I havn't had time to take a look yet.  But steps to reproduce should start from "any workspace" not "this workspace"

wimjongman, 2017-12-19:
Can this be closed?

rtwfroody, 2017-12-19:
Fine by me. I've stopped using Sloeber, largely because of this issue. I couldn't confirm that it's fixed even if you did fix it.

closed 2017-12-19 by rtwfroody

wimjongman, 2017-12-19:
Thanks for the uplifting comment.