Sloeber/arduino-eclipse-plugin #861: OTA upload IP parse error

JanZChlumu, 2017-11-07:
Hello I found small bug
In case of traing to flash ESP8266 over OTA
1. select detected module
[image]
2. attempt to flash ESP8266
[image]
script espota.py expected in parametr _-i IP address_ not hostname, that is reason why script failure
- - -
If I try to modify parametr in command shell, it works correctly, see below.

[image]
- - -
Tested on Eclipse Oxygen.1a Release (4.7.1a) 20171005-1200. Arduino plugin V4 (installed via Eclipse Market)

schreibfaul1, 2017-11-13:
I have the same problem, hostnames ends with .local  thats why they are not known by the router. Remove that ".local"  in Build Variables  A.SERIAL.PORT it will work but only in the current project.

jantje, 2017-11-16:
as a workaround try: as portname use the ipadress a space and the dns name

judge2005, 2017-11-30:
Can you connect to esp-83c25e.local using a browser? If not, you need to install an mDNS service such as bonjour.

jantje, 2017-11-30:
The port name is a text box you can enter whatever you want there. No need to change A.SERIAL.PORT to remove local. Do it in the gui
Important is that sloeber will try to upload via usb when there are no spaces in the portname.
If there is a space Sloeber will try to upload over the network.
From my point of view there is no problem in sloeber. The bonjour service finds the service and sloeber tries to use the name given by the bonjour service (if you select that n the combobox) but this tool seems to need a ip adress and not the .local dns name.
If you do not agree please specify what is the problem.

closed 2017-12-11 by jantje