Sloeber/arduino-eclipse-plugin #862: nRF24: Can't compile

steand, 2017-11-07:
I have try to compile  `Getting Started example sketch for nRF24L01+` from the [nRF24] (https://github.com/nRF24/RF24) for a Arduino nano (ATmege368P).
I got a include Error:
```
19:28:10 **** Incremental Build of configuration Release for project RF24Test ****
"C:\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\.ino.cpp'
'Starting C++ compile'
"C:\sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\cores\arduino" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\variants\eightanaloginputs" -I"C:\sloeber\arduinoPlugin\libraries\RF24\1.3.0" -I"C:\sloeber\arduinoPlugin\libraries\RF24\1.3.0\utility" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\libraries\SPI" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\libraries\SPI\src" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\.ino.cpp"  -o  ".ino.cpp.o"
'Finished building: ..\.ino.cpp'
' '
'Building file: C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\libraries\SPI\src\SPI.cpp'
'Starting C++ compile'
"C:\sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\cores\arduino" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\variants\eightanaloginputs" -I"C:\sloeber\arduinoPlugin\libraries\RF24\1.3.0" -I"C:\sloeber\arduinoPlugin\libraries\RF24\1.3.0\utility" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\libraries\SPI" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\libraries\SPI\src" -MMD -MP -MF"libraries\SPI\src\SPI.cpp.d" -MT"libraries\SPI\src\SPI.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\libraries\SPI\src\SPI.cpp"  -o  "libraries\SPI\src\SPI.cpp.o"
'Finished building: C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\libraries\SPI\src\SPI.cpp'
' '
'Building file: C:\sloeber\arduinoPlugin\libraries\RF24\1.3.0\utility\SPIDEV\compatibility.c'
'Starting C compile'
"C:\sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-gcc" -c -g -Os -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -flto -fno-fat-lto-objects -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR   -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\cores\arduino" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\variants\eightanaloginputs" -I"C:\sloeber\arduinoPlugin\libraries\RF24\1.3.0" -I"C:\sloeber\arduinoPlugin\libraries\RF24\1.3.0\utility" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\libraries\SPI" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.20\libraries\SPI\src" -MMD -MP -MF"libraries\RF24\utility\SPIDEV\compatibility.c.d" -MT"libraries\RF24\utility\SPIDEV\compatibility.c.o" -D__IN_ECLIPSE__=1 "C:\sloeber\arduinoPlugin\libraries\RF24\1.3.0\utility\SPIDEV\compatibility.c"  -o  "libraries\RF24\utility\SPIDEV\compatibility.c.o"
In file included from C:\sloeber\arduinoPlugin\libraries\RF24\1.3.0\utility\SPIDEV\compatibility.c:2:0:
C:\sloeber\arduinoPlugin\libraries\RF24\1.3.0\utility\SPIDEV\compatibility.h:17:22: fatal error: sys/time.h: No such file or directory
 #include <sys/time.h>
                      ^
compilation terminated.
libraries\RF24\utility\SPIDEV\subdir.mk:31: recipe for target 'libraries\RF24\utility\SPIDEV\compatibility.c.o' failed
make: *** [libraries\RF24\utility\SPIDEV\compatibility.c.o] Error 1

19:28:14 Build Finished (took 3s.749ms)
```

With the original Arduino SDK it will compile error free.
My feeling is, that in the Arduino SDK only the code for Arduino are compiled;
In the Eclipse IDE the full code. (RPi, Linux, ....)

Thanks for Help.

steand, 2017-11-08:
It has no priority. I have changed to [mirf](https://github.com/aaronds/arduino-nrf24l01) and this work fine. :-)
But I am still interested on RF24 - it has more features.

jantje, 2017-11-08:
I confirm this to be a bug in Sloeber. This is the first time I see this and it is linked to the arduino library specification 1.0 So I would like to add that the RF24 usage of utility folder is "unique".
The library specification at https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification states:

> For backward compatibility with Arduino 1.0.x, the library author may opt to place source code into the root folder, instead of the folder called src. In this case the 1.0 library format is applied and the source code is searched from the library root folder and the utility folder, for example:
> 
> Servo/Servo.h
> Servo/Servo.cpp
> Servo/utility/ServoTimers.h
> Servo/utility/ServoTimers.cpp
> 
> This will allow existing 1.0.x libraries to compile under 1.5.x+ as well and vice-versa. If a library only needs to run on 1.5.x+, we recommend placing all source code in the src/ folder. If a library requires recursive compilation of nested source folders, its code must be in the src/ folder (since 1.0.x doesn’t support recursive compilation, backwards compatibility wouldn’t be possible anyway)

CDT (by default) does recursive compilation so Sloeber also recursively compiles the utility folder which is not compliant with the text above.

Workaround is to exclude the subfolders of utility.
To do so enter the exclusion filter `libraries/?*/utility/*/**`  in the project properties->C/C++ General->paths and symbols->source location->edit filter->add.
Select OK->OK

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

steand, 2017-11-09:
@jantje Thanks.

To compile the nFR24 I have to exclude:
````libraries/?*/pyRF24/*/**
libraries/?*/examples_linux/*/**
````
also.  
This workaround is fine for me.

jantje, 2017-11-09:
During testing I fund this filter to be better
`libraries/?*/utility/*/*`
I'm closing release 4.2 this month and I want this one added as it is a small "safe" change.

closed 2017-11-21 by jantje