Sloeber/arduino-eclipse-plugin #863: An error has ocurred. See the log file.

TrippyLighting, 2017-11-15:
I posted this on the Arduino Forum as well but have received no response.

On my iMac I currently use Slober 3.1 (ancient) and an even more ancient version 1.30 of Teensyduino.

I installed Sloeber 4.1 (stable) on my new MacBook Pro and the latest Java JDK 9,
which I realize is one step ahead of what is documented on the sloeber.io website, which is 8 but I've installed higher java versions before and it's never been a problem.

I can start Sloeber and it asks for a work space directory, but when I accept the default it responds with the error message:

`An error has ocurred. See the log file`

Here's the log file output:

```
!ENTRY org.eclipse.e4.ui.workbench 4 0 2017-11-12 17:56:29.231
!MESSAGE FrameworkEvent ERROR
!STACK 0
java.lang.NoClassDefFoundError: javax/annotation/PreDestroy
	at org.eclipse.e4.core.internal.di.InjectorImpl.disposed(InjectorImpl.java:426)
	at org.eclipse.e4.core.internal.di.Requestor.disposed(Requestor.java:154)
	at org.eclipse.e4.core.internal.contexts.ContextObjectSupplier$ContextInjectionListener.update(ContextObjectSupplier.java:78)
	at org.eclipse.e4.core.internal.contexts.TrackableComputationExt.update(TrackableComputationExt.java:111)
	at org.eclipse.e4.core.internal.contexts.TrackableComputationExt.handleInvalid(TrackableComputationExt.java:74)
	at org.eclipse.e4.core.internal.contexts.EclipseContext.dispose(EclipseContext.java:176)
	at org.eclipse.e4.core.internal.contexts.osgi.EclipseContextOSGi.dispose(EclipseContextOSGi.java:106)
	at org.eclipse.e4.core.internal.contexts.osgi.EclipseContextOSGi.bundleChanged(EclipseContextOSGi.java:139)
	at org.eclipse.osgi.internal.framework.BundleContextImpl.dispatchEvent(BundleContextImpl.java:903)
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230)
	at org.eclipse.osgi.framework.eventmgr.ListenerQueue.dispatchEventSynchronous(ListenerQueue.java:148)
	at org.eclipse.osgi.internal.framework.EquinoxEventPublisher.publishBundleEventPrivileged(EquinoxEventPublisher.java:213)
	at org.eclipse.osgi.internal.framework.EquinoxEventPublisher.publishBundleEvent(EquinoxEventPublisher.java:120)
	at org.eclipse.osgi.internal.framework.EquinoxEventPublisher.publishBundleEvent(EquinoxEventPublisher.java:112)
	at org.eclipse.osgi.internal.framework.EquinoxContainerAdaptor.publishModuleEvent(EquinoxContainerAdaptor.java:156)
	at org.eclipse.osgi.container.Module.publishEvent(Module.java:476)
	at org.eclipse.osgi.container.Module.doStop(Module.java:634)
	at org.eclipse.osgi.container.Module.stop(Module.java:498)
	at org.eclipse.osgi.container.SystemModule.stop(SystemModule.java:202)
	at org.eclipse.osgi.internal.framework.EquinoxBundle$SystemBundle$EquinoxSystemModule$1.run(EquinoxBundle.java:165)
	at java.base/java.lang.Thread.run(Thread.java:844)
Caused by: java.lang.ClassNotFoundException: javax.annotation.PreDestroy cannot be found by org.eclipse.e4.core.di_1.6.1.v20160712-0927
	at org.eclipse.osgi.internal.loader.BundleLoader.findClassInternal(BundleLoader.java:410)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:372)
	at org.eclipse.osgi.internal.loader.BundleLoader.findClass(BundleLoader.java:364)
	at org.eclipse.osgi.internal.loader.ModuleClassLoader.loadClass(ModuleClassLoader.java:161)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:496)
	... 21 more
```

wimjongman, 2017-11-15:
Stock Sloeber 4.1 will not work with java 9 because it is not based on the latest Eclipse but you can get it running by following these instructions [1].

Alternatively you can install CDT (Eclipse C tooling) Oxygen 1a [2] and install the Sloeber plugins [3]

or just install a Sloeber SDK from the nightly [4]

[1] https://waynebeaton.wordpress.com/2017/09/25/running-eclipse-ide-on-java-9/
[2] https://www.eclipse.org/downloads/packages/eclipse-ide-cc-developers/oxygen1a
[3] http://eclipse.baeyens.it/nightly.php
[4] https://oss.sonatype.org/content/repositories/snapshots/com/github/brodykenrick/arduino-eclipse-plugin/io.sloeber.product/4.1.0-SNAPSHOT/

TrippyLighting, 2017-11-19:
I finally got it to work by installing the nightly Sloeber 4.2. However, I had to uninstall Java 9 (after installing JDK 8) for that to work, otherwise I still did receive the same error message.
Thanks for the help.

wimjongman, 2017-11-19:
Did you install the plugins in your own Eclipse? Sloeber IDE ships with its own JRE. It should never give problems.

TrippyLighting, 2017-11-19:
No, I downloaded the nightly product. This is also on a new Mac Book Pro with a basically out-of-the-box fresh installation of macOS Sierra now High Sierra. So there has not been any previous installation of Java.

I was not aware that this should not be necessary.  I was the first user that used the Arduino Eclipse paging  with Teensy boards and had to install  Java with previous versions of the Arduino Eclipse plugins. So It seemed just natural to me having to do it ;-)

wimjongman, 2017-11-19:
Ah you are running on mac. I'm not sure if the mac distribution includes the jre. However, what worries  me is that you are not able to run the nightly on java 9. I will look into that.

TrippyLighting, 2017-11-19:
Should we re-open the issue ?

wimjongman, 2017-11-19:
Yes

wimjongman, 2017-11-19:
Are you getting the same issue or a different issue? My tests show that 4.1 can run on java 9 with the caveats I published on my first comment.

However, I do run in an issue caused by an old gson library. Are you looking at the same one?

[image]

java.lang.NoClassDefFoundError: java/sql/Time
	at com.google.gson.Gson.<init>(Gson.java:230)
	at com.google.gson.Gson.<init>(Gson.java:174)
	at io.sloeber.core.managers.Manager.loadPackage(Manager.java:250)
	at io.sloeber.core.managers.Manager.loadJson(Manager.java:241)
	at io.sloeber.core.managers.Manager.loadJsons(Manager.java:172)
	at io.sloeber.core.managers.Manager.startup_Pluging(Manager.java:89)
	at io.sloeber.core.Activator$2.run(Activator.java:198)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)

wimjongman, 2017-11-19:
Sloeber 4.1 will not work with java 9.
Sloeber nightly 4.2 before november 18 2017 will work on java 9 if you add the following line to the -vmargs section of the file sloeber-ide.ini:

> --add-modules=ALL-SYSTEM

I will add this setting to the default settings so that future nightlies are fine.

wimjongman, 2017-11-19:
Nightlies should be running fine on java 9 from this point forward

closed 2017-11-19 by wimjongman

jantje, 2017-11-19:
Thanks Wim :-)

TrippyLighting, 2017-11-19:
I can confirm that this works with the nightly. However, there is some "stuff" left that can easily confuse users.

1. Installing the JRE is not enough one has to install the JDK! The documentation says "you're safe" but at least on macOS it's always been a MUST.
2. When a stable version 4.1 is downloaded the splash screen shows "Sloeber Beryllium 4.1". When one downloads a nightly the file name still references 4.1, for example `io.sloeber.product-4.1.0-20171110.002936-136-macosx.cocoa.x86_64.tar.gz` but the splash screen reads "Sloeber Beryllium 4.**2**"

Now I've only one problem left. I've been able to compile older projects, but they show a lot of unresolved references of libraries. I've come across this before but don't remember what the fix was. 

Anyway, thanks again for the help!

TrippyLighting, 2017-11-19:
Found it! Everything works now.

jantje, 2017-11-19:
The 4.2 version thing is because we're in the process of releasing 4.2. The splash has been updated before the version numbers (I did that 18 Nov so it is in the build of 19 Nov).

This makes that the latest nightly is from **Nov 19** and is located at `https://oss.sonatype.org/content/repositories/snapshots/com/github/brodykenrick/arduino-eclipse-plugin/io.sloeber.product/`**4.2.0**`-SNAPSHOT/`
Sloeber.io points to `https://oss.sonatype.org/content/repositories/snapshots/com/github/brodykenrick/arduino-eclipse-plugin/io.sloeber.product/`**4.1.0**`-SNAPSHOT/`  for the  nightly. Here you can find the latest 4.1 nightly **Nov 10**

So basically the version you downloaded `io.sloeber.product-**4.1.0**-2017**1110**.002936-136-macosx.cocoa.x86_64.tar.gz` does not contain the fix Wim has provided :-s
It however does contain a newer eclipse version which solves other java 9 issues.

Note that the version you now have supports auto library install. Make sure you're cdt settings are correct or you may install plenty of unwanted libraries. This is why you get the warning screen if the cdt settings are set on "maximal discovery" and not on "strict code discover".

Would you be willing to share what was needed to fix?

jantje, 2017-11-19:
As to java JRE versus JDK
As to licences we can package the jre but not the jdk.
For windows the jre has been sufficient but I can't speak for mac.
The nightly does not contain the jre. I can't even recall how I added the jre's.

TrippyLighting, 2017-11-19:
To get Sloeber nightly to install I needed to add:
`--add-modules=ALL-SYSTEM`
to the -vmargs section of the file sloeber-ide.ini as suggested by @wimjongman 

The issue with the unresolved references seems to stem from indexer issues. When I install sloeber I get this warning:

```
The indexer option "index source files not included in the build" must be off in windows->preferences->C/C++->indexer 
and
The indexer option "index unused headers" must be off in windows->preferences->C/C++->indexer 

Sloeber might still function but if you get strange results you know where to look.
Do not create an issue if you see this!!! 
```

So I disabled these and while my code compiled (have not tied uploading to a Teensy yet) there are a lot of unresolved references.

When I re-enable the settings these unresolved references disappear, but therefore I get the warning message again. Catch 22 ?

As to JRE vs. JDK, I've always had to install the JDK on macOS. I understand the licensing issue. It might be interesting to ask other ac users for their experience and if this is consistent to update the documentation.

jantje, 2017-11-19:
>So I disabled these and while my code compiled (have not tried uploading to a Teensy yet) there are a lot of unresolved references.

This is strange. Have you tried to rebuild the index?

>to the -vmargs section of the file sloeber-ide.ini as suggested by @wimjongman

This explains as Wim his fix is exactly doing this.

jantje, 2017-11-19:
@TrippyLighting 
I think that  if the issue you have with the indexer is real it is a showstopper for release 4.2.
Can you make sure the indexer does not resolve?
Note that while the indexer is rebuilding there will be unresolved items. Give the indexer some time.
Here are the settings as they should be. The marked ones are the not default ones.
[image]

TrippyLighting, 2017-11-20:
Now that the indexer had some time it seems to work OK.
I do have a minor difference in my settings compared to your screenshot above.
You have "Use active build configuration" enabled. My default is "Use build configuration specified in the project's indexer settings"
 Here is how my settings look at the moment.

[image]

jantje, 2017-11-20:
That is a relieve:-)
The "build configuration for the indexer" is only important if you use multiple configurations.