Sloeber/arduino-eclipse-plugin #864: Changing COM-Port destroys esp8266-Code with <timelib.h>

hfuerst, 2017-11-15:
I did the following steps:
1. extend existing code with Time-Library  #include <TimeLib.h> &  #include <WiFiUdp.h>
2. Verify this Code -> OK
3. Upload this code to ESP # 1 at port COM17 -> everything works
4. Changing Port to COM20 for uploading to ESP # 2
5. Try to Upload -> Code Errors
6. Changing Port back to COM17 for uploading to ESP # 1
7. Try to Upload -> Code Errors
8. now 2 Ways a (with Port COM17) or b with Port COM20) ->
8a. delete the include, new code and remove the Lib (with sloeber-Tool) the Errors are gone, also redoing Step 1 to 3 works.
8b. delete the include, new code and remove the Lib (with sloeber-Tool) the Errors are still here.

9. went back to existing Code (with TortoisHG) Changing Port to COM 20 then redoing the Steps 1-3 the Error is gone, Step 4 changing PORT (now to COM17) - It's all the same....

____________________________________________________________________________________
Errors:
Description	Resource	Path	Location	Type

dereferencing pointer to incomplete type	time.c	/Hasenzaun_Server_AJAX/core/core	line 89	C/C++ Problem
make: *** [core\time.c.o] Error 1	Hasenzaun_Server_AJAX		 	C/C++ Problem
recipe for target 'core\time.c.o' failed	subdir.mk	/Hasenzaun_Server_AJAX/Release/core/core	line 419	C/C++ Problem
unknown type name 'clockid_t'	time.c	/Hasenzaun_Server_AJAX/core/core	line 77	C/C++ Problem
unknown type name 'suseconds_t'	time.c	/Hasenzaun_Server_AJAX/core/core	line 27	C/C++ Problem
unknown type name 'time_t'	time.c	/Hasenzaun_Server_AJAX/core/core	line 26	C/C++ Problem
unknown type name 'time_t'	time.c	/Hasenzaun_Server_AJAX/core/core	line 32	C/C++ Problem
unknown type name 'time_t'	time.c	/Hasenzaun_Server_AJAX/core/core	line 39	C/C++ Problem
unknown type name 'time_t'	time.c	/Hasenzaun_Server_AJAX/core/core	line 46	C/C++ Problem
unknown type name 'time_t'	time.c	/Hasenzaun_Server_AJAX/core/core	line 85	C/C++ Problem
unknown type name 'time_t'	time.c	/Hasenzaun_Server_AJAX/core/core	line 92	C/C++ Problem
unknown type name 'time_t'	time.c	/Hasenzaun_Server_AJAX/core/core	line 107	C/C++ Problem
unknown type name 'time_t'	time.c	/Hasenzaun_Server_AJAX/core/core	line 112	C/C++ Problem
____________________________________________________________________________________

Best regards

Hartmut Fürst

hfuerst, 2017-11-15:
This Timelib:
https://github.com/PaulStoffregen/Time

hfuerst, 2017-11-16:
Today I noticed that the error occurs not only when changing the port, but also when cleaning up the project.

hfuerst, 2017-11-16:
Now I found the way to the closed issues here at github (I'm new here).

The workaround at 
https://github.com/Sloeber/arduino-eclipse-plugin/issues/851
see
https://github.com/Sloeber/arduino-eclipse-plugin/issues/825
does not work !
Same Errors

Thx
H.Fuerst

jantje, 2017-11-16:
There is probably a unwanted library in your project.
Each time you change something (like the arduino project properties) a scan is done for "missing libraries" and if found they are auto imported (by default; you can disable this in the arduino preferences)
Once they have been imported they will not be "auto deleted".
After aplying the actions for #851 and #825 you can delete all (unwanted) libraries in the [project]->libraries folder and then have sloeber re-Import the libraries.

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

closed 2017-11-23 by jantje