Sloeber/arduino-eclipse-plugin #865: Using an ino file named differently from the project

winginitau, 2017-11-16:
Apologies if this is an eclipse noob question......

I have an existing project that I manage with git. Structure:

git/FodderFactory/ff_arduino/ff.ino    //main arduino file
git/FidderFactory/ff_simulator/ff_simulator.cpp    //equivalent linux gcc main file
git/FodderFactory/src    //all of the *cpp and *h include files used in both the arduino and linux builds

In eclipse with sloeber nightly build plugin:

New Arduino Sketch
Project:
ff-workspace/ff_arduino

delete ff_arduino.ino  ///created with the project

File/Import/Arduino/"Import a folder containing source code file in the current project"
point to: git/FodderFactory/ff_arduino

*folder imported and ff.ino is found* **though it is called ff.ino _not_ ff_arduino.ino

add ....src as a folder link under c/c++ source location. Add sames for includes, link and assembly.

All is well, however there is "nothing to build".

How do I tell the plugin to build for the imported ff.ino rather than the (now deleted) ff_arduino.ino?

Many thanks
Brendan

jantje, 2017-11-16:
Sloeber currently only looks for .ino files in the root folder.

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

winginitau, 2017-11-16:
Many thanks. Yes it works if I drag the file to root in the eclipse project explorer (even though it has a name different from the project) and it doesn't change the underlying file location managed by git. All good.

closed 2017-11-16 by winginitau

wimjongman, 2017-11-17:
>  and it doesn't change the underlying file location managed by git. All good.

Is this what you want? Don't you want to manage the Sloeber project with Git?

jantje, 2017-11-17:
@wimjongman I think @winginitau moved a link and as such didn't change the actual location of the file so there is not necessarily version control impact ( some eclipse config file will have changed but those do not have to be checked in)

jantje, 2017-11-17:
@winginitau thanks for becoming a patron.

wimjongman, 2017-11-17:
I don't think so. When @winginitau creates a project like he did, it is inside his eclipse workspace which does not point back to his repos working directory.

jantje, 2017-11-17:
@wimjongman 
Not the best way to create a link to git versions code but the action

> File/Import/Arduino/"Import a folder containing source code file in the current project"
point to: git/FodderFactory/ff_arduino

males links and does not copy code around

wimjongman, 2017-11-17:
> makes links and does not copy code around

Yes, but the following action:

> .. it works if I drag the file to root in the eclipse project explorer

moves that file to the workspace out of his git directory

jantje, 2017-11-17:
> moves that file to the workspace out of his git directory

@winginitau 
If Wim says so I would double check. Note that you can have 2 links to the ino file so you can have a compatible setup.

winginitau, 2017-11-17:
@wimjongman you are quite right. In error, I initially thought that it only changed the project explorer location, but on further inspection it has indeed moved the file out of the repo and into the project directory. So I've fixed the problem by manually making a sym link:

@mbpumdev:~/ff-workspace/ff_arduino$ ln -s /home/brendan/git/FodderFactory/ff_arduino/ff.ino .

@jantje it would be nice to have sloeber find ino files deeper in the tree, however given that ino projects really only need a single ino file, it makes sense to have it in the root for simplicity - and a sym link is an easy workaround for git integration. 

@wimjongman the import process actually creates an ff_arduino folder under project root, so although ti makes a link, it is not in the root to be found by sloeber. 

How would you suggest a better method? (I'm very new to both eclipse and git so still feeling my way...) 

See https://github.com/winginitau/FodderFactory for how I'm organised. I have 3x binaries to build (1x avr, 2x linux). I currently do that using 3x projects and manually modifying src/ff_sys_config.h changing pre-proc #defines to get it to pick up the right code for each target. Is there a better way?

jantje, 2017-11-18:
> @jantje it would be nice to have sloeber find ino files deeper in the tree, however given that ino projects really only need a single ino file, it makes sense to have it in the root for simplicity - and a sym link is an easy workaround for git integration.

This has come up before and I agree. See #747 Though I don't think this is urgent.
Note that (as you clearly know about c/cpp) you can easily rename your .ino to .cpp (though that would make it arduino ide incompatible)

I looked at your project and it  doesn't make sense to me. This is normal as I don't know what fodder is nor barley sprouts and there are no eclipse configuration files.
The missing eclipse configuration files is because you used
File/Import/Arduino/"Import a folder containing source code file in the current project"
and not 
File/Import/Git/projects from Git/existing local repository

> How would you suggest a better method? (I'm very new to both eclipse and git so still feeling my way...)

This used to be my day time job. There is no quick over the internet answer to this question. There is also no "one and only way". It is a search and getting people involved is very likely to be a "pay them" activity.
A search is however lots easier when you know the basics so maybe this will interest you:

https://www.patreon.com/posts/are-you-in-video-15434866