Sloeber/arduino-eclipse-plugin #867: I am working on a unit test of arduino code using sloeber. I am not sure what I am doing wrong.

Danial-Zafar, 2017-11-20:
[image]
[image]

jantje, 2017-11-20:
>I am not sure what I am doing wrong.

So am I

Danial-Zafar, 2017-11-20:
so kindly tell me why I'm facing these errors I have done everything that is mentioned in that video https://www.youtube.com/watch?v=d-Vt21BIHl0&t=312s 
why 'm facing these errors
it gives me an error for ...\Arduino eclipse IDE\sloeber\arduinoPlugin\tools\make\make": Launching failed
there is no \tools\make\make,  in my directory

wimjongman, 2017-11-20:
Have you read the FAQ? If you do not want to read it then you can support the project on Patreon which enables you to ask questions without reading the documentation ;)

(Hint 1: no spaces in the pathname)

[1] http://eclipse.baeyens.it/faq.shtml
[2] https://www.patreon.com/jantje

Danial-Zafar, 2017-11-21:
(Hint 1: no spaces in the pathname) I remove the space but still facing two error :( 

make: *** [sloeber.ino.cpp.o] Error 1
recipe for target 'sloeber.ino.cpp.o' failed	subdir.mk

ichigo92, 2017-11-22:
I am also facing similar issue with `make`, i have added details on issue #867.

wimjongman, 2017-11-22:
@Danial-Zafar what version are you using. It looks like a very old version of sloeber. Please upgrade. 

[image]

wimjongman, 2017-11-22:
Closed. See #870

closed 2017-11-22 by wimjongman

Danial-Zafar, 2017-11-22:
I'm using 4.2 now before that I was using 4.0 and 4.1